openrazer/openrazer #521: nothing provides typelib(Gdk) needed by openrazer-daemon-2.2.0-2.1.noarch

fiercebrake, 2018-02-02:
Greetings,

I am on fedora 27, just followed the instryctions to install the openrazer, but i get the error

 nothing provides typelib(Gdk) needed by openrazer-daemon-2.2.0-2.1.noarch

is there any particular repo i need to enable besides http://download.opensuse.org/repositories/hardware:razer/Fedora_27/hardware:razer.repo ?

thanks for your help!

z3ntu, 2018-02-02:
No it's fine. I added the dependency yesterday or so because that dependency was missing on opensuse tumbleweed and the build with that also went fine on Fedora so that's why I pushed it. Guess I'll remove it again for Fedora tomorrow hopefully.

z3ntu, 2018-02-02:
Ref #519

fiercebrake, 2018-02-02:
Thanks!

ill try again tomorrow or day after

closed 2018-02-04 by z3ntu