openrazer/openrazer #522: Possible kernel memory corruption / driver crash

PetteriAimonen, 2018-02-03:
After installing openrazer-kernel-modules-dkms   2.2.0~ubuntu16.04.1 on kernel 4.10.0-42-generic with Razer Ornata, my computer has started crashing every few days. The timing matches and it's previously been very stable, but there is nothing directly pointing to openrazer so this is just a hunch for now. I'll uninstall the driver and monitor what happens, but here is a report in case someone else hits this.

Log from crash:
<pre>
Feb  3 18:47:55 oddish kernel: [335093.779817] BUG: unable to handle kernel paging request at fffff395e9215aa0
Feb  3 18:47:55 oddish kernel: [335093.779958] IP: kfree+0x4f/0x170
Feb  3 18:47:55 oddish kernel: [335093.780008] PGD 0 
Feb  3 18:47:55 oddish kernel: [335093.780010] 
Feb  3 18:47:55 oddish kernel: [335093.780071] Oops: 0000 [#1] SMP
Feb  3 18:47:55 oddish kernel: [335093.780118] Modules linked in: uas usb_storage cdc_acm md4 nls_utf8 cifs fscache msr razerkbd(OE) hid_generic usbhid hid ccm pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) rfcomm vboxdrv(OE) bnep bbswitch(OE) binfmt_misc rtsx_usb_ms asus_nb_wmi memstick asus_wmi sparse_keymap nls_iso8859_1 mxm_wmi snd_hda_codec_hdmi snd_hda_codec_realtek intel_rapl snd_hda_codec_generic x86_pkg_temp_thermal intel_powerclamp snd_hda_intel uvcvideo kvm_intel snd_hda_codec kvm snd_hda_core snd_hwdep btusb irqbypass videobuf2_vmalloc snd_pcm btrtl videobuf2_memops videobuf2_v4l2 btbcm crct10dif_pclmul videobuf2_core btintel videodev crc32_pclmul bluetooth media ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd snd_seq_midi snd_seq_midi_event glue_helper snd_rawmidi cryptd arc4 intel_cstate intel_rapl_perf
Feb  3 18:47:55 oddish kernel: [335093.781034]  snd_seq iwldvm snd_seq_device mac80211 joydev input_leds snd_timer serio_raw snd iwlwifi soundcore mei_me cfg80211 mei lpc_ich shpchp int3403_thermal mac_hid acpi_als wmi int3402_thermal kfifo_buf int340x_thermal_zone industrialio int3400_thermal asus_wireless acpi_thermal_rel asus_fan(OE) coretemp parport_pc ppdev lp parport autofs4 btrfs xor raid6_pq rtsx_usb_sdmmc rtsx_usb i915 i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops psmouse drm ahci libahci fjes video [last unloaded: nvidia]
Feb  3 18:47:55 oddish kernel: [335093.781651] CPU: 0 PID: 700 Comm: rtsx_usb_ms_1 Tainted: P     U  W  OE   4.10.0-42-generic #46~16.04.1-Ubuntu
Feb  3 18:47:55 oddish kernel: [335093.781777] Hardware name: ASUSTeK COMPUTER INC. UX32VD/UX32VD, BIOS UX32VD.214 01/29/2013
Feb  3 18:47:55 oddish kernel: [335093.781883] task: ffff8a4f9b1a0000 task.stack: ffff98d1823b0000
Feb  3 18:47:55 oddish kernel: [335093.781966] RIP: 0010:kfree+0x4f/0x170
Feb  3 18:47:55 oddish kernel: [335093.782021] RSP: 0018:ffff98d1823b3a60 EFLAGS: 00010282
Feb  3 18:47:55 oddish kernel: [335093.782094] RAX: fffff395e9215a80 RBX: fffff1f74856a140 RCX: 0000000000000000
Feb  3 18:47:55 oddish kernel: [335093.782188] RDX: 000075b280000000 RSI: fffff1f7447f0100 RDI: fffff1f74856a140
Feb  3 18:47:55 oddish kernel: [335093.782281] RBP: ffff98d1823b3a78 R08: 000000001fc04101 R09: 0000000180150009
Feb  3 18:47:55 oddish kernel: [335093.782374] R10: 0000019ea9215a80 R11: 0000000000000000 R12: 0000000000000001
Feb  3 18:47:55 oddish kernel: [335093.782466] R13: ffffffffb36a01ab R14: 0000000000000000 R15: fffff1f74856a140
Feb  3 18:47:55 oddish kernel: [335093.782562] FS:  0000000000000000(0000) GS:ffff8a4faee00000(0000) knlGS:0000000000000000
Feb  3 18:47:55 oddish kernel: [335093.782666] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Feb  3 18:47:55 oddish kernel: [335093.782744] CR2: fffff395e9215aa0 CR3: 000000020d209000 CR4: 00000000001406f0
Feb  3 18:47:55 oddish kernel: [335093.782837] Call Trace:
Feb  3 18:47:55 oddish kernel: [335093.782887]  usb_control_msg+0xeb/0x130
Feb  3 18:47:55 oddish kernel: [335093.782947]  hub_ext_port_status+0x96/0x140
Feb  3 18:47:55 oddish kernel: [335093.783010]  usb_port_resume+0x38e/0x760
Feb  3 18:47:55 oddish kernel: [335093.783073]  ? usb_runtime_suspend+0x70/0x70
Feb  3 18:47:55 oddish kernel: [335093.783135]  generic_resume+0x15/0x20
Feb  3 18:47:55 oddish kernel: [335093.783193]  usb_resume_both+0xf4/0x130
Feb  3 18:47:55 oddish kernel: [335093.789491]  usb_runtime_resume+0x1a/0x20
Feb  3 18:47:55 oddish kernel: [335093.795634]  __rpm_callback+0xc4/0x200
Feb  3 18:47:55 oddish kernel: [335093.801709]  ? usb_runtime_suspend+0x70/0x70
Feb  3 18:47:55 oddish kernel: [335093.807784]  rpm_callback+0x24/0x80
Feb  3 18:47:55 oddish kernel: [335093.813797]  ? usb_runtime_suspend+0x70/0x70
Feb  3 18:47:55 oddish kernel: [335093.819784]  rpm_resume+0x49f/0x6a0
Feb  3 18:47:55 oddish kernel: [335093.825746]  rpm_resume+0x2d9/0x6a0
Feb  3 18:47:55 oddish kernel: [335093.831661]  ? pick_next_task_fair+0x3d6/0x4d0
Feb  3 18:47:55 oddish kernel: [335093.837527]  ? __switch_to+0x23c/0x530
Feb  3 18:47:55 oddish kernel: [335093.843436]  rpm_resume+0x2d9/0x6a0
Feb  3 18:47:55 oddish kernel: [335093.849178]  ? try_to_del_timer_sync+0x5a/0x80
Feb  3 18:47:55 oddish kernel: [335093.854897]  rpm_resume+0x2d9/0x6a0
Feb  3 18:47:55 oddish kernel: [335093.860570]  ? del_timer_sync+0x50/0x50
Feb  3 18:47:55 oddish kernel: [335093.866212]  __pm_runtime_resume+0x4e/0x80
Feb  3 18:47:55 oddish kernel: [335093.871835]  rtsx_usb_detect_ms_card+0x76/0x120 [rtsx_usb_ms]
Feb  3 18:47:55 oddish kernel: [335093.877441]  kthread+0x109/0x140
</pre>

Which seems to happen in kfree() in the usb driver for my card reader. Perhaps some kind of heap corruption?

PetteriAimonen, 2018-03-26:
Yeah, correlation seems pretty strong, multiple crashes per day with openrazer installed and none in two months without it.

z3ntu, 2018-03-26:
When the crashes happen do you interact with the driver at all? Or just when the driver is loaded? Can you reproduce the issue at all or is it just these random crashes?

PetteriAimonen, 2018-03-26:
It seems to happen after resume-from-ram or after connecting or disconnecting the keyboard, but not immediately, but after a few minutes of delay. I cannot reproduce it easily, even if I suspend and resume many times it doesn't always occur, but in normal usage (a few disconnects per day) it occurs daily.

Symptoms are that first all input devices stop working but clock on the screen still runs, then after a few seconds the clock also stops and I have to reboot, after which I find the crash message in /var/log.

terrycain, 2018-03-28:
uninstall it and see if it stops

PetteriAimonen, 2018-03-28:
Yes, no crashes since I uninstalled openrazor in February.

closed 2018-08-25 by stale[bot]