openrazer/openrazer #523: Black Widow Ultimate 2016 accidentally hit macro key now it just keeps blinking

krutoileshii, 2018-02-04:
Just need some help. Accidentally hit macro key and now can't disable it. the blinking red light is annoying . How do I turn it off.

Emerald-Night, 2018-02-04:
Navigate to the directory that your device is detected by openrazer in a terminal. The directory is something like /sys/bus/hid/drivers/razerkbd/

There are probably going to be multiple folders in there. run "ls <folder name>" on each of them until you see a file with the word "macro" in it; cd into the directory with that file. After that, I think you run "echo 0 > macro_file".

That should turn off the light.

krutoileshii, 2018-02-04:
Awesome thank you. Dang thing was driving me nuts.

closed 2018-02-04 by krutoileshii

Emerald-Night, 2018-02-04:
No worries at all.