openrazer/openrazer #525: device.brightess is randomly 0

Rasie1, 2018-02-07:
Razer Blade Stealth Late 2017, Manjaro with kernel 4.15

When I read `device.brightness` from library in python, it's `0.0` sometimes, which is incorrect

Here is script for testing:
```
#!/usr/bin/python
from openrazer.client import DeviceManager
from openrazer.client import constants as razer_constants
import time

device_manager = DeviceManager()
device_manager.sync_effects = False
device = device_manager.devices[0]
device.brightness = 40

while True:
    print("Step:")
    print("Before             : " + str(device.brightness))
    device.brightness = 40
    print("After (immediately): " + str(device.brightness))
    time.sleep(0.1)
    print("After (0.1 of wait): " + str(device.brightness))
    time.sleep(1.0)
```

Here is example output when I run it:
```
Before             : 40.0
After (immediately): 40.0
After (0.1 of wait): 40.0
Step:
Before             : 0.0
After (immediately): 40.0
After (0.1 of wait): 40.0
Step:
Before             : 40.0
After (immediately): 40.0
After (0.1 of wait): 40.0
Step:
Before             : 40.0
After (immediately): 0.0
After (0.1 of wait): 40.0
Step:
Before             : 40.0
After (immediately): 40.0
After (0.1 of wait): 40.0
Step:
Before             : 40.0
After (immediately): 40.0
After (0.1 of wait): 40.0
Step:
Before             : 40.0
After (immediately): 40.0
After (0.1 of wait): 40.0
Step:
Before             : 40.0
After (immediately): 40.0
After (0.1 of wait): 0.0
Step:
Before             : 40.0
After (immediately): 40.0
After (0.1 of wait): 40.0
```

Rasie1, 2018-02-13:
Also, sometimes when I set keyboard backlighting through "device.fx.advanced.draw()", not all keys are getting updated. This happens rarily, not as frequently as issue above, but I think it's related somehow

z3ntu, 2018-03-30:
Yes, the Razer devices sometimes don't accept the commands as they're busy or something.

Womagrid, 2018-07-14:
Was having the same problem with my Razer Diamondback Chroma mouse...

Increasing RAZER_MOUSE_WAIT_MIN_US to 1000 in razermouse_driver.h appears to have fixed it.  I also increased RAZER_MOUSE_WAIT_MAX_US to 1200.

Note that the equivalent constants in other files may need to be adjusted depending on the type of device.

I think this is also the cause of #604

Rasie1, 2018-07-27:
I ended up requesting brightness 3 times and selecting the most common value

muesli, 2021-06-17:
Fwiw, this issue seems to have disappeared on my system with recent releases. Can anyone else still reproduce this?

Aciz, 2021-12-14:
I am still experiencing this on 3.2.0 with Viper Ultimate, pretty much every time I wake up my computer from sleep, the mouse LED is off.