openrazer/openrazer #526: Brightness for Abyssus V2 resets after reboot

tmitzka, 2018-02-14:
I use the Razer mouse Abyssus V2 with the [Polychromatic](https://github.com/polychromatic/polychromatic) applet on Linux Mint 18.3. Adjusting effects, brightness and color works fine, but after rebooting, the brightness is reset to the default (?) value. For the logo, I set brightness to zero and it works. After a reboot the logo is glowing again.

In the log file I didn't see anything suspicious.

My user is in the plugdev group. Still, `openrazer-daemon -Fv` returns `Unable to lock on the pidfile.`

Here is the output of `python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`:

`[<RazerMouse PM1636H03803860>]`

And `systemctl --user status openrazer-daemon.service` shows this:

`openrazer-daemon.service
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)`

What other outputs or attempts might help?

z3ntu, 2018-02-14:
`Unable to lock on the pidfile.` means that the daemon is already running.
The python output with `<RazerMouse ...` means that the daemon recognized your device.
The systemctl output is like this because you are technically running Ubuntu 16.04.

About the brightness resetting, I am not sure of the logic there in the daemon but apparently the daemon always sets the brightness to 100 on startup.

tmitzka, 2018-02-14:
Can I tell it not to do that? Maybe through a config file?

z3ntu, 2018-02-15:
https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/mouse.py#L821-L824 That's the code that does that.

tmitzka, 2018-02-24:
Thanks! I commented these lines out, now it works as expected.

closed 2018-02-24 by tmitzka