openrazer/openrazer #527: UEFI Support

Spikeophant, 2018-02-18:
I recently swapped to linux on my work workstation and am using UEFI secure boot.  I have gotten the openrazer drivers working without issue and would be happy to contribute back scripts to create and load keys into the MOK, sign any existing openrazer .ko files, as well as scripts that will resign upon DKMS compile if there is any interest?

z3ntu, 2018-02-18:
A bunch of people had secure boot turned on but nobody really saw this as an issue yet ^^ but maybe it'd be better if you would put those script in a seperate git repository where we can link to if the scripts are generic for all kinds of dkms modules.

Spikeophant, 2018-02-18:
Okay I will do that, everything I did was generic enough to be used on any DKMS module.  I had googled around on my issue and the only posts I saw were people saying turn off UEFI secure boot, so I figured I would contribute back how to make it work.

z3ntu, 2018-03-30:
I'll close this issue as it's technically not an issue on our side. :) We can link to your scripts though if you want to

closed 2018-03-30 by z3ntu