openrazer/openrazer #528: Doesnt find any razer devices on Manjaro Linux

Moquo, 2018-02-20:
Hello,
openrazer doesnt find any devices on Manjaro (Arch) Linux. Im using Polychromatic controller. I installed everything and started the daemon, but it doesnt work too.
Thanks for you help.

Rasie1, 2018-02-20:
This might be the same issue https://github.com/openrazer/openrazer/issues/520

Moquo, 2018-02-20:
sudo dkms install openrazer-driver/1.0.0; sudo modprobe razerkbd
```
Error! echo
Your kernel headers for kernel 4.14.19-1-MANJARO cannot be found at
/usr/lib/modules/4.14.19-1-MANJARO/build or /usr/lib/modules/4.14.19-1-MANJARO/source.
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.14.19-1-MANJARO
```
python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
```
[]
```

z3ntu, 2018-02-20:
Please read the first message and install your kernel headers.

Moquo, 2018-02-20:
The problem is that I already installed the kernel headers (`sudo pacman -S linux-headers`)

z3ntu, 2018-02-20:
If the kernel headers aren't at `/usr/lib/modules/$(uname -r)/build` (even after a reboot) then there's something wrong with the installation.

closed 2018-07-28 by z3ntu