openrazer/openrazer #529: openrazer-daemon fail to start - RazerBladeStealthLate2016

lecler-i, 2018-02-21:
Hello,

I'm having an issue.. I can't have my openrazer-daemon to start anymore. Here are the logs : 

```
2018-02-21 11:42:53 | razer                          | INFO     | Found device.0: 0003:1532:0220.0004
2018-02-21 11:42:53 | razer.device0                  | INFO     | Initialising device.0 RazerBladeStealthLate2016
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 126, in run_daemon
    test_dir=args.test_dir)
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/daemon.py", line 107, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/daemon.py", line 398, in _load_devices
    razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/keyboards.py", line 50, in __init__
    super(_RippleKeyboard, self).__init__(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/keyboards.py", line 19, in __init__
    super(_MacroKeyboard, self).__init__(*args, **kwargs)
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/device_base.py", line 84, in __init__
    DBusService.__init__(self, self.BUS_PATH, object_path)
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/dbus_services/service.py", line 87, in __init__
    super(DBusService, self).__init__(bus_object, object_path)
  File "/usr/lib/python3.6/site-packages/dbus/service.py", line 455, in __init__
    validate_object_path(object_path)
ValueError: Invalid object path '/org/razer/device/empty (NULL)': contains invalid character ' '

Stopping daemon.
Exception ignored in: <bound method RazerDevice.__del__ of RazerBladeStealthLate2016:empty (NULL)>
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/device_base.py", line 421, in __del__
    self.close()
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/device_base.py", line 335, in close
    self._close()
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/keyboards.py", line 55, in _close
    super(_RippleKeyboard, self)._close()
  File "/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/keyboards.py", line 39, in _close
    self.key_manager.close()
AttributeError: 'RazerBladeStealthLate2016' object has no attribute 'key_manager'
  
```

Kernel module is loaded and initialized without problem.
Anyone had a similar issue ?

z3ntu, 2018-02-21:
Please apply the patch https://github.com/openrazer/openrazer/commit/af2db82a4f8ae6156a34b3fb1fc262147fa7e799 to your installation. Also it's a kind-of duplicate of #192

lecler-i, 2018-02-21:
Partially fixed : 

https://github.com/lecler-i/openrazer/commit/9b3999cadd311ed0824a6705694b502d2e3f0d0b

Daemon is now working, using my testing script : 

```
from openrazer.client import DeviceManager
from openrazer.client import constants as razer_constants

device_manager = DeviceManager()

print("Found {} Razer devices".format(len(device_manager.devices)))

device_manager.sync_effects = False
for device in device_manager.devices:
    dict = {}
    rows, cols = device.fx.advanced.rows, device.fx.advanced.cols
    for row in range(rows):
        for col in range(cols):
            device.fx.advanced.matrix[row, col] = (0, 255, 0)
            device.fx.advanced.draw()
            device.fx.advanced.matrix[row, col] = (0, 0, 0)

    print(device.name + ":")
```

lecler-i, 2018-02-21:
`polychromatic` is still not working but at least it conect to the daemon and list my device.

All this is pretty weird cause it was working all perfectly before last week... and I didn't change much into y config >.<

closed 2018-02-21 by z3ntu