openrazer/openrazer #531: unable to use my naga chroma mouse on Debian

ti1336, 2018-02-21:
Hi,

i'm trying to use my naga chroma on my office computer with an MX Linux distribution, but i'm unable to make it works... 
I've trying some solved issues, but i'm still doing something wrong :/ 
It's important for me, my left arm is paralized and this mouse is very helpfull for me. 

thank you in advance 

PS : sorry for my bad english

z3ntu, 2018-02-21:
The output of the following commands please:
`sudo dkms install openrazer-driver/1.0.0; sudo modprobe razerkbd`
`python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`

ti1336, 2018-02-21:
`Module openrazer-driver/1.0.0 already installed on kernel 4.13.0-1-amd64/x86_64`
`modprobe: FATAL: Module razer not found in directory /lib/modules/4.13.0-1-amd64`
`[<RazerMouse PM1729H00901230>]`

z3ntu, 2018-02-21:
Except that you copy pasted the first command wrong 😉 , it seems fine.
What isn't working?

ti1336, 2018-02-21:
When i'm trying to open custom editor he said to me : device type not implemented
[image]

z3ntu, 2018-02-21:
This is an issue for the RazerGenie repository, not the openrazer one :slightly_smiling_face: => https://github.com/z3ntu/RazerGenie/

closed 2018-02-21 by z3ntu

ti1336, 2018-02-21:
Oh ok sorry ! Thank you for your help :)