openrazer/openrazer #532: OpenRazer Python Library Could Not Be Imported

dboybash, 2018-02-22:
I'm trying to use Polychromatic Controller with a Chroma Ornata keyboard.  I've followed the documentation for installation but I'm getting a message saying Polychromatic requires the OpenRazer Python library, but it could not be imported.  Not sure what to do/where to go next as I'm an Ubuntu noob.  Any assistance would be much appreciated.  Thank you kindly in advance.

lah7, 2018-02-22:
Presumably, if you have followed the install instructions, `python3-openrazer` should be installed, which is needed by the daemon and the front-end application. Installing `openrazer-meta` would've included this too.

If you open a **Terminal** and run this command - what is the output?

    python3 -c "import sys; print('\n'.join(sys.path))"

On Ubuntu 16.04, this would output something similar to this:
```
/usr/lib/python35.zip
/usr/lib/python3.5
/usr/lib/python3.5/plat-x86_64-linux-gnu
/usr/lib/python3.5/lib-dynload
/usr/local/lib/python3.5/dist-packages
/usr/lib/python3/dist-packages
```

If this looks wildly different, this might be similar to https://github.com/polychromatic/polychromatic/issues/137

dboybash, 2018-02-22:
Thank you for the response.  The resulting output is 
/usr/lib/python35.zip
/usr/lib/python3.5
/usr/lib/python3.5/plat-x86_64-linux-gnu
/usr/lib/python3.5/lib-dynload
/usr/local/lib/python3.5/dist-packages
/usr/lib/python3/dist-packages

I'm assuming the ppa installation was successful.  Wondering if I've missed a step someplace because I'm still getting the Daemon Not Installed screen.  I was under the impression that following those steps would install/configure everything properly before installing Polychromatic.  Are there other checks I should/can do to make sure no issues are resulting from OpenRazer?  Again, thank you in advance!!

[image]

lah7, 2018-02-23:
Which version of Ubuntu are you using?

Let's try re-installing everything, in case the packages were misconfigured along the way - please paste the output of:
```
sudo apt remove *openrazer*
sudo add-apt-repository ppa:openrazer/stable
sudo apt update
sudo apt install openrazer-meta
```

**`openrazer-meta`** will install `openrazer-daemon`, `openrazer-doc`, `openrazer-kernel-modules-dkms`  and `python3-openrazer` (the last one is what appears to be missing)

Once OpenRazer is installed, you can try running the daemon and install the front-end again:
```
openrazer-daemon
sudo apt install polychromatic
```

dboybash, 2018-02-23:
Thank you for your response! That resolved my issue!!  Much appreciated!!

closed 2018-02-23 by z3ntu