openrazer/openrazer #533: Razer Blade 2017 Support

michaelrubino27, 2018-02-22:
Do you guys plan on supporting the 2017 Razer Blade?

z3ntu, 2018-02-22:
What's the USB ID of the device?

michaelrubino27, 2018-02-24:
Bus 001 Device 005: ID 1532:0224 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0224 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

total 0
drwxr-xr-x 2 root root 220 Feb 24 17:55 .
drwxr-xr-x 4 root root 600 Feb 24 17:55 ..
lrwxrwxrwx 1 root root   9 Feb 24 17:55 usb-Microsoft_Microsoft®_Nano_Transceiver_v1.0-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Feb 24 17:55 usb-Microsoft_Microsoft®_Nano_Transceiver_v1.0-if01-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Feb 24 17:55 usb-Microsoft_Microsoft®_Nano_Transceiver_v1.0-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Feb 24 17:55 usb-Microsoft_Microsoft®_Nano_Transceiver_v1.0-if02-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 Feb 24 17:55 usb-Razer_Razer_Blade-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Feb 24 17:55 usb-Razer_Razer_Blade-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 Feb 24 17:55 usb-Razer_Razer_Blade-if02-event-mouse -> ../event12
lrwxrwxrwx 1 root root   9 Feb 24 17:55 usb-Razer_Razer_Blade-if02-mouse -> ../mouse1