openrazer/openrazer #534: Python framework: how do I get started?

VikingScientist, 2018-02-22:
How do I get started using the python library to program my keyboard? In particular what is the `serial` argument to `openrazer.client.devices.keyboard.RazerKeyboard`

If I try to run the following piece of code

``` python
from openrazer.client.devices import keyboard
kb = keyboard.RazerKeyboard('razerblackwidowultimate2016')
```
Then I get the following exception:
```
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/openrazer/client/devices/keyboard.py", line 12, in __init__
    super(RazerKeyboard, self).__init__(serial, vid_pid=vid_pid, daemon_dbus=daemon_dbus)
  File "/usr/lib/python3/dist-packages/openrazer/client/devices/__init__.py", line 30, in __init__
    self._name = str(self._dbus_interfaces['device'].getDeviceName())
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "getDeviceName" with signature "" on interface "razer.device.misc" doesn't exist
```
Are there any documentations or getting started guides available? Could someone give by the python3 code I need to light my 'K' key to 50% and the 'home'-key to 100% on my Blackwidow Ultimate 2016 keyboard?

lah7, 2018-02-22:
The correct usage would be:

```
import openrazer.client
devman = openrazer.client.DeviceManager()
devices = devman.devices()
kbd = devman.devices[0]   // List of all devices, assuming keyboard is first one.
```

When using the Python console, you can poke `kbd` with tab to see all available functions or variables, such as getting the name of the device:

```
>>> kbd.name
'Razer BlackWidow Chroma'
```

When it comes to brightness, at a hardware level, it's controlled by a number between 0 and 100.

However, if you are to apply different **static** colours to the keyboard, you'll set the RGB in matrix and then send that to the device.

```
total_rows = device.fx.advanced.rows
total_cols = device.fx.advanced.cols
```

For instance, this will set the `ESC` key to green:
```
kbd.fx.advanced.matrix[0, 1] = [0, 255, 0]
kbd.fx.advanced.draw()
```

To emulate a different brightness, you'll need to set the key to a darker shade of green, like `[0, 30, 0]`

> Are there any documentations or getting started guides available?

I believe @terrycain was writing documentation quite some time ago, I'm not sure what's happened to it.

z3ntu, 2018-02-23:
Also we have examples in the [`examples/`](https://github.com/openrazer/openrazer/tree/master/examples) directory.

redaxmedia, 2018-03-09:
Please write some API documentation / references.

z3ntu, 2018-03-09:
Python API documentation was started on the branch [`feature_docs`](https://github.com/openrazer/openrazer/tree/feature_docs).

closed 2018-08-27 by stale[bot]

inftp, 2018-10-11:
Can this be re-opened? It seems the docs never progressed

z3ntu, 2018-10-12:
Contributions welcome to that branch.