openrazer/openrazer #535: sysfs: cannot create duplicate filename matrix_effect_static

enyone, 2018-02-23:
```
[13184.785879] usb 5-4: new full-speed USB device number 4 using xhci_hcd
[13184.935758] usb 5-4: New USB device found, idVendor=1532, idProduct=021f
[13184.935764] usb 5-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[13184.935768] usb 5-4: Product: Razer Ornata
[13184.935770] usb 5-4: Manufacturer: Razer
[13184.938087] input: Razer Razer Ornata as /devices/pci0000:00/0000:00:1d.0/0000:05:00.0/0000:06:01.0/0000:08:00.0/0000:09:02.0/0000:0c:00.0/usb5/5-4/5-4:1.0/0003:1532:021F.000A/input/input25
[13184.998679] hid-generic 0003:1532:021F.000A: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Ornata] on usb-0000:0c:00.0-4/input0
[13185.000566] input: Razer Razer Ornata as /devices/pci0000:00/0000:00:1d.0/0000:05:00.0/0000:06:01.0/0000:08:00.0/0000:09:02.0/0000:0c:00.0/usb5/5-4/5-4:1.1/0003:1532:021F.000B/input/input26
[13185.058434] hid-generic 0003:1532:021F.000B: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer Ornata] on usb-0000:0c:00.0-4/input1
[13185.059925] input: Razer Razer Ornata as /devices/pci0000:00/0000:00:1d.0/0000:05:00.0/0000:06:01.0/0000:08:00.0/0000:09:02.0/0000:0c:00.0/usb5/5-4/5-4:1.2/0003:1532:021F.000C/input/input27
[13185.060845] hid-generic 0003:1532:021F.000C: input,hidraw6: USB HID v1.11 Mouse [Razer Razer Ornata] on usb-0000:0c:00.0-4/input2
[13185.686993] input: Razer Razer Ornata as /devices/pci0000:00/0000:00:1d.0/0000:05:00.0/0000:06:01.0/0000:08:00.0/0000:09:02.0/0000:0c:00.0/usb5/5-4/5-4:1.1/0003:1532:021F.000B/input/input28
[13185.763199] input: Razer Razer Ornata as /devices/pci0000:00/0000:00:1d.0/0000:05:00.0/0000:06:01.0/0000:08:00.0/0000:09:02.0/0000:0c:00.0/usb5/5-4/5-4:1.0/0003:1532:021F.000A/input/input29
[13185.806452] sysfs: cannot create duplicate filename '/devices/pci0000:00/0000:00:1d.0/0000:05:00.0/0000:06:01.0/0000:08:00.0/0000:09:02.0/0000:0c:00.0/usb5/5-4/5-4:1.2/0003:1532:021F.000C/matrix_effect_static'
[13185.806472] ------------[ cut here ]------------
[13185.806489] WARNING: CPU: 0 PID: 5914 at /build/linux-3RM5ap/linux-4.14.13/fs/sysfs/dir.c:31 sysfs_warn_dup+0x51/0x60
[13185.806704] CPU: 0 PID: 5914 Comm: razer_mount Tainted: G        W  O    4.14.0-0.bpo.3-amd64 #1 Debian 4.14.13-1~bpo9+1
[13185.806706] Hardware name: LENOVO 20HD0001MX/20HD0001MX, BIOS N1QET67W (1.42 ) 10/03/2017
[13185.806718] RIP: 0010:sysfs_warn_dup+0x51/0x60
[13185.806743] Call Trace:
[13185.806758]  sysfs_add_file_mode_ns+0x10f/0x170
[13185.806767]  razer_kbd_probe+0x1f6/0xba0 [razerkbd]
[13185.806779]  hid_device_probe+0x9c/0xf0 [hid]
[13185.806787]  driver_probe_device+0x293/0x440
[13185.806792]  bind_store+0x105/0x150
[13185.806799]  kernfs_fop_write+0x109/0x1a0
[13185.806811]  vfs_write+0xb0/0x190
[13185.806817]  SyS_write+0x52/0xc0
[13185.806823]  do_syscall_64+0x7c/0x120
[13185.806829]  entry_SYSCALL64_slow_path+0x25/0x25
[13185.806928] ---[ end trace fa98133891d0987d ]---
[13185.861951] razerkbd 0003:1532:021F.000B: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Ornata] on usb-0000:0c:00.0-4/input1
[13185.921971] razerkbd 0003:1532:021F.000A: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer Ornata] on usb-0000:0c:00.0-4/input0
```

z3ntu, 2018-02-23:
Thanks for the bug report. I will push a fix shortly.

closed 2018-02-23 by z3ntu

enyone, 2018-02-23:
OMG, what a response speed :1st_place_medal:
Thanks.

commit 0961f275 (2018-02-23): Add missing 'break' for Razer Ornata in razerkbd

Fixes #535

Caused by 1d94785