openrazer/openrazer #536: How to mock the DeviceManager

redaxmedia, 2018-02-23:
Hello guys,

I'm currently refactoring Chroma Feedback to unit testing. I have to mock the DeviceManager somehow or let's say mock the Device List.

1. The core is passing `device_manager.devices` to a method - [Link](https://github.com/redaxmedia/chroma-feedback/blob/master/src/core.py#L44)
1. The device handler is processing it using `device.name` and `device.fx` - [Link](https://github.com/redaxmedia/chroma-feedback/blob/master/src/device.py#L4)
1. I used just the Device List to be able to write a unit test when no real device is available:

### tests/test_device.py

```
import src.device as device

mock =\
[
	{
		'name': 'Razer Chroma',
		'fx': None
	}
]

def test_process_device_passed():
	data = device.process_device(mock, 'passed')
	assert 'Setting Razer Chroma to build passed' in data['message'][0]
	assert data['status'] == 'passed'
```

### error

```
data = [{'fx': None, 'name': 'Razer Chroma'}], status = 'passed'

    def process_device(data, status):
        message = []
    
        # process data
    
        for device in data:
>               if status == 'passed' and static(device.fx, [0, 255, 0]):
E     AttributeError: 'dict' object has no attribute 'fx'

src/device.py:10: AttributeError
```

I guess this is a noob question, but I write Python since a couple of months only - sorry guys :-)

z3ntu, 2018-02-23:
I think you are looking for https://stackoverflow.com/a/11532079/3527128

redaxmedia, 2018-02-23:
Well, I'd like to mock the list and not the class...

redaxmedia, 2018-02-24:
I made it using MagicMock... thank you guys

closed 2018-02-24 by redaxmedia