openrazer/openrazer #538: Kernel driver fails to install

Tank-Missile, 2018-02-24:
Linux distribution: Artix Linux (Arch derivative) 
Kernel version: 4.15.5-1 (also attempted on lts 4.14.21-1)

Error from make.log:
> ::^[[32m Compiling OpenRazer kernel modules^[[0m
> ========================================
> make -C /lib/modules/4.15.5-1-ARTIX/build SUBDIRS=/var/lib/dkms/openrazer-driver/1.0.0/build/driver modules
> make[1]: Entering directory '/usr/lib/modules/4.15.5-1-ARTIX/build'
> gcc: error trying to exec 'cc1': execvp: No such file or directory
> gcc: error trying to exec 'cc1': execvp: No such file or directory
> Makefile:934: *** "Cannot generate ORC metadata for CONFIG_UNWINDER_OR/var/lib/dkms/openrazer-driver/1.0.0/build/make.logC=y,
> please install libelf-dev, libelf-devel or elfutils-libelf-devel".  Stop.
> make[1]: Leaving directory '/usr/lib/modules/4.15.5-1-ARTIX/build'
> make: *** [Makefile:27: driver] Error 2

z3ntu, 2018-02-25:
It seems that your gcc installation is broken.
You can put this in a file called `main.c`
```c
#include <stdio.h>

int main() {
    printf("hello world");
}
```
and then run `gcc main.c -o main` from that folder. That has to work before you can try compiling anything else. I would ask in the Artix Linux forums/IRC/whatever for help.

z3ntu, 2018-02-25:
Do you have the base-devel package group installed?

Tank-Missile, 2018-02-25:
Yes.

Tank-Missile, 2018-03-06:
Yes.

z3ntu, 2018-03-06:
Output of `/usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/cc1 -version` ?

Tank-Missile, 2018-03-06:
```
GNU C11 (GCC) version 7.3.0 (x86_64-pc-linux-gnu)
	compiled by GNU C version 7.3.0, GMP version 6.1.2, MPFR version 4.0.0, MPC version 1.1.0, isl version isl-0.18-GMP

warning: MPFR header version 4.0.0 differs from library version 4.0.1.
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
```

z3ntu, 2018-03-06:
Do any other dkms packages (e.g. virtualbox-host-dkms) work on your system?

Tank-Missile, 2018-03-06:
Well, this is indeed strange. virtualbox-host-dkms fails to install as well. Perhaps something is wrong with my dkms installation? dkms is version is 2.5-1, which is the same version Arch Linux currently has. 

```
DKMS make.log for vboxhost-5.2.8_OSE for kernel 4.15.7-1-ARTIX (x86_64)
Tue Mar  6 16:20:57 EST 2018
make: Entering directory '/usr/lib/modules/4.15.7-1-ARTIX/build'
gcc: error trying to exec 'cc1': execvp: No such file or directory
gcc: error trying to exec 'cc1': execvp: No such file or directory
Makefile:934: *** "Cannot generate ORC metadata for CONFIG_UNWINDER_ORC=y, please install libelf-dev, libelf-devel or elfutils-libelf-$
make: Leaving directory '/usr/lib/modules/4.15.7-1-ARTIX/build
```

z3ntu, 2018-03-06:
If all dkms modules fail, this is not an issue with this project. I would suggest you asking for help in the forum for your specific distro.

closed 2018-03-06 by z3ntu

d0tfile, 2018-03-06:
@Tank-Missile, re-read error description, it may be helpful (please install libelf-dev, libelf-devel or elfutils-libelf-$)