openrazer/openrazer #539: No module named 'openrazer_daemon'

wemmsi, 2018-02-26:
Hello everyone,

since I have installed Python 3.6.4 (I also got Python 2.7 and Python 3/3.5 on my system) the openrazer drivers won't start/work anymore.
If I try to start the openrazer-daemon, the error "No module named 'openrazer_daemon'" is raised.
I'm working on Ubuntu 16.04 and my "python3 -c 'import sys; print(sys.path)'" is:
['', '/usr/local/lib/python36.zip', '/usr/local/lib/python3.6', '/usr/local/lib/python3.6/lib-dynload', '/home/lv/.local/lib/python3.6/site-packages', '/usr/local/lib/python3.6/site-packages']

Kind regards,
Lisa

z3ntu, 2018-02-26:
The Ubuntu package installs the python components into `/lib/python3/dist-packages/openrazer_daemon` so you will have to configure your python executable to look into this path (which the default installation does).

wemmsi, 2018-02-27:
I will try to configure it!

wemmsi, 2018-03-08:
I fixed it with setting:
``#!/usr/bin/python3``
in /usr/bin/openrazer-daemon!
Thanks!

closed 2018-03-08 by wemmsi