openrazer/openrazer #540: Support for Lancehead wireless mouse?

delooper, 2018-02-26:
Greetings,
 
I installed openrazer tonight and noticed it does not appear to support the Lancehead wireless mouse.  Is there anything I could do to help create support for this mouse?  

I have a functioning install of the driver -- it notices my Naga Hex V2 just fine.

delooper, 2018-02-26:
lsusb -d 1532: -v

Bus 003 Device 002: ID 1532:005a Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        32
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x005a 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

delooper, 2018-02-26:
ls -lah /dev/input/by-id
total 0
drwxr-xr-x 2 root root 140 Feb 26 01:59 .
drwxr-xr-x 7 root root 720 Feb 26 01:59 ..
lrwxrwxrwx 1 root root  10 Feb 26 01:51 usb-Chicony_Electronics_Co.__Ltd._Integrated_Camera-event-if00 -> ../event22
lrwxrwxrwx 1 root root   9 Feb 26 01:52 usb-Razer_USB_Dongle_for_Razer_Lancehead-event-mouse -> ../event5
lrwxrwxrwx 1 root root   9 Feb 26 01:52 usb-Razer_USB_Dongle_for_Razer_Lancehead-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Feb 26 01:52 usb-Razer_USB_Dongle_for_Razer_Lancehead-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Feb 26 01:52 usb-Razer_USB_Dongle_for_Razer_Lancehead-mouse -> ../mouse0

delooper, 2018-02-26:
Would any more images help?

[image]
[image]
[image]
[image]
[image]

ranobrega, 2018-12-02:
Hello. If more information is needed, please let me know. I use a Lancehead on Ubuntu Linux and look forward to support from openrazer.

z3ntu, 2018-12-03:
I still need packet captures of the device, see https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

nightsky30, 2019-05-11:
I recently added the bits for partial support of the Lancehead TE (not the side LEDs) in my fork.  I have a [test branch](https://github.com/nightsky30/openrazer/tree/lancehead-0x005A) with what I believe is needed for the same level of partial support for the Lancehead if it is similar enough to the TE version and you wish to test your device on it.

I don't wish to detract from the request for packets.  There are still features not supported which could use some help from those with the devices.

christianhujer, 2019-06-25:
Here's some more information.
For me, the Lancehead has device id `1532:0059` (maybe because it's connected wired).

Here's the `lsusb` call:
```
christian.hujer@Nelkinda-Blade-Stealth-2:~$ sudo lsusb -v -d 1532:0059
Bus 005 Device 004: ID 1532:0059 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0059 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Lancehead
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

```

scott-hyperdrive, 2019-06-30:
> I recently added the bits for partial support of the Lancehead TE (not the side LEDs) in my fork. I have a [test branch](https://github.com/nightsky30/openrazer/tree/lancehead-0x005A) with what I believe is needed for the same level of partial support for the Lancehead if it is similar enough to the TE version. Anyone is welcome to test that if they want.
> 
> I don't wish to detract from the request for packets. There are still features not supported which could use some help from those with the devices.

Can you confirm my installation method for this branch: Installed openrazer-meta & polychromatic from README.md link. sudo make from branch directory. Asking because I'm not seeing any of the additional support listed when starting openrazer-daemon -Fv

wwwjunkboy, 2019-07-06:
Hello I am just bumping this posting. I'd still like support for this mouse. If I can be of any help please let me know. I'm new to linux but am happy to follow instructions on providing info from the mouse.

nightsky30, 2019-07-22:
@wwwjunkboy @christianhujer @ranobrega @delooper 

Would any of you be willing to test my openrazer and razergenie test branches based on the work done for the Lancehead TE?

https://github.com/nightsky30/openrazer/tree/feature_lancehead_0x0059_0x005a
https://github.com/nightsky30/RazerGenie/tree/side-leds

I'm betting the devices are almost identical, but I don't own the hardware to test the normal Lancehead.  I **think** I've added everything required.  If you do feel like testing, please be very thorough.  Especially with the battery stuff as the TE didn't have a wireless component.

djmulder, 2019-07-24:
@nightsky30 nope not working sadly. 

Logs neatly say it's there tho:
`2019-07-24 13:54:06 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2019-07-24 13:54:06 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)`

I'm curious about this tho: 

`2019-07-24 13:54:06 | razer                          | INFO     | Found device.0: 0003:1532:005A.0003
2019-07-24 13:54:06 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1532:005A.0003/device_type, file is not owned by plugdev`

djmulder, 2019-07-24:
tho I must add some notes: 
- it seems (also in polychromatic) that the all the settings reset to default after exiting (even going to a new window and back resets) .. buuut in the mouse the settings remains. So it's more a graphical glitch. 

But that still makes it hard to really work with the DPI.

Also it reports the battery life to 0 all the time it seems.

nightsky30, 2019-07-24:
> Logs neatly say it's there tho:
> 2019-07-24 13:54:06 | razer | DEBUG | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
> 2019-07-24 13:54:06 | razer | DEBUG | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)

Hmmm is it initially detecting it as a Mamba (2012)?

> I'm curious about this tho:
> 
> 2019-07-24 13:54:06 | razer | INFO | Found device.0: 0003:1532:005A.0003
> 2019-07-24 13:54:06 | razer | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1532:005A.0003/device_type, file is not owned by plugdev

Could be #877 

> oops n/m re-connecting device worked perfectly :D thx!

So it is working and detecting as a Lancehead?  Is it working and detected properly both wired and wirelessly from boot without need to plug it in again?  In my experience #877 can sometimes require me to unplug and replug my Lancehead TE after booting.  But barring #877 I'm hoping generally it is detecting it as a Lancehead at boot.

> it seems (also in polychromatic) that the all the settings reset to default after exiting (even going to a new window and back resets) .. buuut in the mouse the settings remains. So it's more a graphical glitch.
> 
> But that still makes it hard to really work with the DPI and yeh colors.. it now nicely breath between blue and purple.. but in the UI it's still on the default (cycle between colors)

Yes, the frontends do not reflect the current mouse settings when they load.  That affects all devices.  I believe that issue is being tracked elsewhere.

With the test branch of RazerGenie:

- Do you have the ability to set effects on all LED lighting sections (scroll, logo, left row, right row)?
- Do you have the ability to set brightness on all LED lighting sections (scroll, logo, left row, right row)?
- Do all available effects function properly in each lighting section?
- Can you set the Polling rate and DPI accurately?

> Also it reports the battery life to 0 all the time it seems.

That is interesting...

Thank you for your help in testing :+1:

djmulder, 2019-07-24:
my apologies, I copy/pasted the wrong lines, it's lancehead in the log as well. 
-
Tests:
I have the ability to set effects on all LED lighting sections idd 
I have the ability to set Brightness on all sections
They seem to function properly
I can set the polling rate (although I have no idea how to test if it really is affected)
and I can set the DPI, this one is very noticeable that it works.
-
Battery life, I disabled in the openrazer config to not give the message anymore, but yeh now I have no indication what it's battery is.

nightsky30, 2019-07-24:
> my apologies, I copy/pasted the wrong lines, it's lancehead in the log as well.
>
> Tests:
> I have the ability to set effects on all LED lighting sections idd
> I have the ability to set Brightness on all [...]

Cool Beans!

> Battery life, I disabled in the openrazer config to not give the message anymore, but yeh now I have no indication what it's battery is.

I made some tweaks to the battery stuff.  Can you update and retest?

djmulder, 2019-07-24:
No message yet, but also nowhere to see the battery life. 

Just noticed something else tho: in RazerGenie there's "Open Custom Editor", when I click that it just says: "device type not implemented, please open an issue in the opengenie repo" (yeh I'm at the wrong repo I know :P but yeh seeing you're active here ;)

nightsky30, 2019-07-24:
> No message yet, but also nowhere to see the battery life.

I guess no message is good assuming you undid the openrazer config change mentioned above.  I don't own any wireless razer devices, so I'm not exactly sure how it is displayed in the GUI normally.

Do you have the Mamba (2012) mentioned earlier, and does that show battery info?

> Just noticed something else tho: in RazerGenie there's "Open Custom Editor", when I click that it just says: "device type not implemented, please open an issue...

Yeah the custom editor for mice is also currently broken and tracked elsewhere.

djmulder, 2019-07-24:
Nah I only have the Oroborous (however you spell it) and it's wireless part isn't supported either 

and polychromatic shows less actually. With Razergenie I can set all the colors, with Polychromatic only the logo and the wheel.

nightsky30, 2019-07-24:
From this [comment](https://github.com/openrazer/openrazer/issues/576#issuecomment-383263693) it looks like it may just be in the daemon logs and screen notifications?

> Nah I only have the Oroborous (however you spell it) and it's wireless part isn't supported either

Sorry to hear that.  Yeah, I didn't have many examples to base my changes off of with the battery stuff.

> and polychromatic shows less actually. With Razergenie I can set all the colors, with Polychromatic only the logo and the wheel.

I've only put the changes in for the left and right side LEDs in openrazer and RazerGenie, and even those pull requests are still pending.  I've not tried modifying Polychromatic yet, but would like to.  I'm not as familiar with the code.

Well if things are working well enough in these test branches and the other pull requests are merged, we could try submitting a PR for the Lancehead afterwards.  Glad things seem to be working generally :)

djmulder, 2019-07-24:
heh I'm already happy it works. Battery would just be gravy :)

idd a PR would be nice.. Perhaps they'll do more with it

wwwjunkboy, 2019-07-25:
I installed razer genie and it sees my razer laptop but still doesn't see the mouse. Do I need to do anything beyond installing razer genie?

nightsky30, 2019-07-25:
> I installed razer genie and it sees my razer laptop but still doesn't see the mouse. Do I need to do anything beyond installing razer genie?

You need to install git, and then clone both forked repositories of openrazer and razergenie mentioned above.  After that, checkout the test branches, build and install/run.

OpenRazer
```
git clone https://github.com/nightsky30/openrazer.git
cd openrazer
git checkout feature_lancehead_0x0059_0x005a
```
Not sure what distro you are using, but with Debian/Ubuntu the following should work from inside the local repo...
```
./scripts/build_debs.sh
sudo dpkg -i ./dist/*.deb
```
Then reboot!

RazerGenie
```
git clone https://github.com/nightsky30/RazerGenie.git
cd RazerGenie
git checkout side-leds
meson build
ninja -C build
./build/src/razergenie
```

nightsky30, 2019-07-31:
If anyone prefers Polychromatic to RazerGenie I've now fixed a few issues in my openrazer test branch that were giving Polychromatic issues.  You'll need to update, rebuild, openrazer.  I also forked and created a test branch for Polychromatic(Stable) that supports the side LEDs.

https://github.com/nightsky30/openrazer/tree/feature_lancehead_0x0059_0x005a
https://github.com/nightsky30/RazerGenie/tree/side-leds
https://github.com/nightsky30/polychromatic/tree/side-leds

Everything should be functional now in Polychromatic, but the layouts are a little wonky.

WhyNotHugo, 2019-09-10:
I've a Lancehead Wireless, but mine does not work at all (eg: moving or clicking does not work).
Its id is `1532:006f`.

What can I contribute to add some support for it?

nightsky30, 2019-09-10:
@WhyNotHugo Testing is definitely appreciated.

> I've a Lancehead Wireless, but mine does not work at all (eg: moving or clicking does not work).
> Its id is 1532:006f.
> 
> What can I contribute to add some support for it?
> 
> Never mind that. It works with one USB adapter, but not another. Super weird.

- Is 0x006f the product ID for the non-functional adapter, perhaps for a different unsupported mouse?
- When it works with the other adapter does it match the product IDs 0x0059 and 0x005a as mentioned by others above?  Just want to verify we are talking about the same device.

> I'd still like to know how I can contribute to getting DPI/acceleration configurable on it, since it's useless for me in the meantime. :)

@djmulder mentioned above the DPI was configurable, though that was before the device classes were modified in the PR.  Perhaps that has something to do with it.

jeaye, 2019-10-19:
Hey folks. I've just been trying my Lancehead and I've built these branches from source. I'm able to tweak my LEDs and DPI in both Polychromatic and RazerGenie, so that's a success.

However, my largest concern is around rebinding the mouse buttons to keys. The actual binding is trivial, using xbindkeys, but the problem I'm seeing is twofold:

* The two buttons on the left side are button 8 and button 9, but the two buttons on the right side are **also** button 8 and button 9
* The two buttons in front of the mouse wheel don't send an event to `ev` and instead just change the mouse's DPI

So, is there a way to differentiate the left and right side buttons? Also, is there a way to detect when the top two buttons in front of the mouse wheel have been pressed?

Finally, @nightsky30, is there anything I can do to help with this?

nightsky30, 2019-10-20:
@jeaye Thank you for testing the LEDs and DPI.  I've experienced the duplicated side buttons on the Lancehead TE.  Not exactly sure what can be done there.  But there appears to be a separate effort into key rebinding of mice and keyboards with #880.

americansoup, 2019-11-01:
When running 'sudo make install' I get the following response: 

"Please do not install the driver using this method. Use a distribution package as it tracks the files installed and can remove them afterwards. If you are 100% sure, you want to do this, find the correct target in the Makefile. 
Exiting."

This makes perfect sense however, I can't get the driver version that supports the Lancehead wireless by using my distro's package manager (portage w/vifino-overlay overlay) and I am not sure what the correct target is from the makefile (install_i_know_what_i_am_doing?)/how to proceed to manually install the drivers.

Thanks ahead of time.

nightsky30, 2019-11-04:
> When running 'sudo make install' I get the following response:
> 
> "Please do not install the driver using this method. Use a distribution package as it tracks the files installed and can remove them afterwards. If you are 100% sure, you want to do this, find the correct target in the Makefile.
> Exiting."
> 
> This makes perfect sense however, I can't get the driver version that supports the Lancehead wireless by using my distro's package manager (portage w/vifino-overlay overlay). I'm assuming using the clean target is what I want to. Is this correct? Are there any things that folks already know that I should be aware of before going this route?
> 
> Thanks ahead of time.

For a testing install it seems you would want the target above that:
```sudo make install_i_know_what_i_am_doing```

nightsky30, 2019-11-04:
> When running 'sudo make install' I get the following response:
>
> "Please do not install the driver using this method. Use a distribution package as it tracks the files installed and can remove them [...]

For a testing install it seems you would want the target above that:
```
sudo make install_i_know_what_i_am_doing
```

americansoup, 2019-11-07:
Thanks for replying. Going to try all and driver_install when I get home tonight. Hopefully that will do the trick.

americansoup, 2019-11-23:
After trying all, and driver_install I can see that it did update the modules but I still don't have any control over the Lancehead. Not sure what to try next...

nightsky30, 2019-11-23:
I've always gone the .deb package route with the daemon by using ./scripts/build_debs.sh and then installing the generated .debs.  But I believe you mentioned you were on another distro.

The makefile lists:
https://github.com/openrazer/openrazer/blob/974cefaac71388ea0c185df5fdf5c093b5651cc4/Makefile#L157
Perhaps try:
`sudo make install_i_know_what_i_am_doing`
 You probably still need the udev and python target installs which that target ends up calling. I imagine the udev_install would be required to update the udev rules file with the Lancehead PID.  And likewise python_library_install for the Lancehead class defined in mouse.py.

Cybiko, 2020-01-02:
Hey all,
Could someone generate a .deb file for me, because I have no idea how to build this (I'm a complete linux noob). 

Thanks!

z3ntu, 2020-01-03:
@Cybiko 
Follow https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint but replace the first two command with the following
```
git clone https://github.com/openrazer/openrazer
cd openrazer/
git fetch origin pull/887/head
git checkout FETCH_HEAD
```
You might get an error message while running the script, when that happens a path to an error log is printed and then you can check that file to see what is wrong - likely missing dependencies which you have to install with apt-get.

nightsky30, 2020-01-11:
Would someone kindly pull and build the latest as outlined by z3ntu above, then test, and verify the mouse and all its various abilities are functioning **wirelessly**.  Please report findings in #887 Thanks

mduft, 2020-01-27:
just chiming in to let you hear that I just got myself a lancehead (2019). if support for this device eventually pops up, that'd be great :)

```
mduft@fril0041 / $ lsusb -v -d 1532:0070

Bus 001 Device 030: ID 1532:0070 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0070 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

nightsky30, 2020-01-28:
@mduft With the Lancehead (2019) version having different device IDs (0070/????) vs (0059/005a) and potentially requiring different support (z3ntu et al. might need pcaps), you may want to open a separate issue to track Lancehead (2019) support so it doesn't get lost in the mix.

z3ntu, 2020-03-14:
@Cybiko @americansoup @jeaye @wwwjunkboy (and others that have this device): Please test #887 so we can get support for this device merged. Thanks!

americansoup, 2020-03-16:
I will definitely test, but I believe that I'm having probs w/my Distro's
DKMS so my results may not be valid.

amit-feldman, 2020-04-11:
the RGB lights only work on scroll wheel and logo, the side rgb's (Main?) remain in color spectrum and can't be modified

nightsky30, 2020-04-11:
> the RGB lights only work on scroll wheel and logo, the side rgb's (Main?) remain in color spectrum and can't be modified

@amit-feldman The stable versions of the frontends don't expose the side LED settings yet.  You're welcome to use z3ntu's [razer_test](https://github.com/z3ntu/RazerGenie/tree/razer_test) branch, or either of my tweaked branches for [RazerGenie](https://github.com/nightsky30/RazerGenie/tree/side-leds) or [Polychromatic](https://github.com/nightsky30/polychromatic/tree/side-leds).

Also, can you verify if the wireless is working well?  If there are any issues with wireless functionality or the battery is not reporting properly I have a possible fix to test.

Thank you for testing.  If we can get it fully tested, then we can get it merged.

amit-feldman, 2020-04-11:
Sure thing!, I'll give it a go and test the wireless as well.
Will report back @nightsky30

amit-feldman, 2020-04-11:
@nightsky30 not workin and I'm not sure what I'm doing wrong, can you share your step-by-step (including commands) to try it out?

amit-feldman, 2020-04-11:
@nightsky30 wireless seems to be working fine, except for the side-leds.

nightsky30, 2020-04-11:
> @nightsky30 not workin and I'm not sure what I'm doing wrong, can you share your step-by-step (including commands) to try it out?

@amit-feldman For my RazerGenie branch:
```
git clone https://github.com/nightsky30/RazerGenie.git
cd RazerGenie
git checkout side-leds
meson build
ninja -C build
./build/src/razergenie
```
And the battery is reporting accurate power percentages?

PR #887 (2020-05-01): Add support for Razer Lancehead
Based on what was done for the Lancehead TE, I've added the pieces required for the Lancehead (0x0059, 0x005a).  From initial testing by a few helpful people, everything seems to be on par with the Lancehead TE regarding detection, dpi, poll rate, brightness, and lighting effects.  This fixes #540.

My only real concern is the battery/charging status support.  I've added what I think is necessary.~~, though I wasn't sure how the battery/charging status is normally shown.  Is it just by screen notifications?~~

~~Definitely a work in progress that could use more  [... cut]
closed 2020-05-01 by z3ntu