openrazer/openrazer #541: Razer Naga Trinity Support

CajetanP, 2018-02-28:
Currently existing Trinity issue seems empty/dead/both so I decided to open up a proper one.

[So, the output of lsusb -d 1532: -v](https://pastebin.com/H1Qdwuz4)
[And ls -lah /dev/input/by-id/](https://pastebin.com/HX9vL2NT)
[Razer Synapse screenshots](https://imgur.com/a/gc4Oo)

Product number: RZ01-02410100
Model number: RZ01-0241

I don't think implementing support for this particular device will be exceptionally difficult since most of the functionality was already implemented for other Naga models. The only new thing is supporting those changing sides but I guess that's feasible too. Let's hope it will work out well!

z3ntu, 2018-02-28:
Please get packet captures of changing dpi, changing light effect (incl brightness) and changing polling rate

manisjp, 2018-03-08:
I will try to get this data for you as I have the same mouse. I have two questions though. Do you want this data from a windows environment using Synapse? If you want this data from within Linux how do I change the colors and polling rates?

z3ntu, 2018-03-08:
Yes, packet captures of what Razer Synapse on Windows is sending to the USB device.

CajetanP, 2018-03-12:
Okay so I got those captures you asked for. I put Wireshark files in a zip [here](https://wetransfer.com/downloads/4ccc67e41a177cfeb5d02f45ff12002620180312155003/79c04b).
In case it comes in handy, I described more or less everything I did in each capture session.

**dpi**
Started with dpi set to 800, then switched it using mouse's dpi buttons like that:
up (1000), up (16000), down (1000), down (800 again).
Mouse brightness was set to off to prevent unnecessary packet flow, light profile was set to Static Red but it doubt it had any effect since the lights were off anyway.

**polling**
Brightness was set to off, started with the polling rate of 1000. Switched to 500, 125, 500 and back to 1000.

**brightness**
Light profile set to Static Red. Started with brightness off, then switched it to on (100%), normal (67%), Dim (33%) and then off again.

**basic_light_effects**
Started with static red set to full brightness. Switched to spectrum cycling, reactive (medium green), clicked a few times, switched to breathing and then back to static.

**adv_light_effects**
Started with full brightness and red-blue fire from chroma studio, switched to wave and then back to fire.

I hope it's of any use, let me know if you need anything else. Sorry for the delay, I've been really busy lately so it was pretty hard to find the time.

z3ntu, 2018-03-12:
Thanks, the packet captures look good. DPI, polling rate and brightness should be able to be implemented easily. Colors and effects might be a bigger problem as Synapse 3 likes to send all effects manually basically.

Anyways, is it possible to configure lighting effects separately on the 3 LEDs (scrolling wheel, logo and keypad)? Or what exactly did you do in the captures?

z3ntu, 2018-03-12:
And please test out the code in the branch `feature_naga_trinity_0x0067`. This version only has support for DPI, polling rate and brightness but please test that by installing that version (if you need help just ask), reboot and then run
```
cd $(dirname /sys/bus/hid/drivers/razermouse/0*/device_serial)
tail device_serial dpi poll_rate matrix_brightness
```

CajetanP, 2018-03-12:
Well basic effect option just sets all three to the same colour & effect so there's not much there. When you use Chroma Studio though, you can apply different colour sets to all 3 LEDs as you wish. In my captures I just set all three to the same one (ex. Red-Blue fire effect or full spectrum wave).

I see, but is implementing effects more or less possible or not? On my main linux I'm just stuck on spectrum cycling all the time which is my least favourite thing so that's not great. Though no pressure, I hope we'll get there eventually. Hopefully I'll be able to understand what your code does well enough to actually contribute at some point but it might take a while since it's my first contact with drivers etc. Can you recommend any useful resources btw?

z3ntu, 2018-03-12:
Yeah please make a capture where you set each LED individually (and brightness individually if possible). For implementing effects we can hope that the firmware in the device still accepts the "old-school" commands because I don't really know how the command spamming should be handled ^^

CajetanP, 2018-03-18:
I cloned the repository, installed with
`make install_i_know_what_i_am_doing`,
rebooted and tried to run
`cd $(dirname /sys/bus/hid/drivers/razermouse/0*/device_serial)`
but I got 
`zsh: no matches found: /sys/bus/hid/drivers/razermouse/0*/device_serial`.
I'm not sure how this works exactly, what should I do differently?

z3ntu, 2018-03-18:
@CajetanP Which distro are you on? Installing it via the Makefile is a bad idea (which is why the target is named like this...) as prefixes etc will probably be wrong.

CajetanP, 2018-03-18:
Yeah that was expectable, so how exactly do I go about installing it?
I'm on Antergos (Arch).

z3ntu, 2018-03-18:
Clone the openrazer-git package from the AUR and edit the source array in the PKGBUILD to include `#branch=branchname` so it'll be `source=("$pkgbase::git+https://github.com/openrazer/openrazer.git#branch=feature_naga_trinity_0x0067")`

CajetanP, 2018-03-18:
So I cloned the package with `git clone https://aur.archlinux.org/openrazer-git.git`, adjusted the PKGBUILD as you said and installed with `makepkg -si`. After rebooting mouse lights are on, it shows up in lsusb but neither the cursor nor the buttons seem to respond to any action. Also, trying that cd command still gives the same "no matches found" message.

z3ntu, 2018-03-18:
Anything in `sudo journalctl -b | grep -i razer_mount` ?

CajetanP, 2018-03-18:
Yes, [that](https://pastebin.com/Adc3ZVrF).

z3ntu, 2018-03-18:
Please run `sudo modprobe razermouse && sudo bash -x /lib/udev/razer_mount razermouse 0003:1532:0067.0007`

CajetanP, 2018-03-18:
[Here's the output.](https://pastebin.com/aRJQ1zvW)

z3ntu, 2018-03-18:
`ls -lad /sys/bus/hid/drivers/hid-generic/0003:1532:0067*`

CajetanP, 2018-03-18:
`zsh: no matches found: /sys/bus/hid/drivers/hid-generic/0003:1532:0067*`

z3ntu, 2018-03-18:
`ls -la /sys/bus/hid/drivers/razer*/` ?

CajetanP, 2018-03-18:
[Here](https://pastebin.com/m5NdFktF)

z3ntu, 2018-03-18:
Maybe `printf '%s' "0003:1532:0067.0007" > /sys/bus/hid/drivers/razermouse/bind`

CajetanP, 2018-03-18:
How should I run it? Without sudo  or with sudo for one of the commands it says "permission denied", with sudo before printf and before /sys/ it doesn't return anything at all.

z3ntu, 2018-03-18:
Actually as root user (otherwise the command would have to be restructured).
Afterwards check if there's something new in `/sys/bus/hid/drivers/razermouse/`.

CajetanP, 2018-03-18:
As root it's just `bash: printf: write error: No such device`.  
Not sure if it helps but lsusb line reads `Bus 001 Device 008: ID 1532:0067 Razer USA, Ltd `

z3ntu, 2018-03-18:
Please rebuild the package with the latest commit and try again.

CajetanP, 2018-03-18:
Just to be sure, how do I check if the latest commit was installed properly?

z3ntu, 2018-03-18:
The version of the packages should contain 59a89e7 (so probably `2.2.2.rSOMETHING.59a89e7-1`)

CajetanP, 2018-03-18:
I think we've finally made some progress!
[Here's the output of "tail device serial (...)".](https://pastebin.com/XgFGfcLC)
The dpi part changes accordingly if I switch profiles using mouse buttons and its value is correct.
Polling rate is also correct but I can't switch it in linux to see about changing.

z3ntu, 2018-03-18:
Looks good!
Can you try if running `echo -n "255" | sudo tee /sys/bus/hid/drivers/razermouse/0*/matrix_brightness` changes the brightness? (you can change the 255 to something between 0 and 255 to also check values inbetween)

CajetanP, 2018-03-18:
Yeah it works fine. 0 turns the lights off completely, 255 turns it to the maximum brightness, values in between work accordingly. Though in that tail output, matrix_brightness field remains at 0 no matter what.

z3ntu, 2018-03-18:
Our "docs" say for matrix_brightness that it's write-only so it might be that the devices don't store that internally. https://github.com/openrazer/openrazer/wiki/Using-the-mouse-driver#matrix_brightness---sets-mouse-and-dock-brightness-write-only

CajetanP, 2018-03-18:
I see, well I guess what matters is that it's changeable.

CajetanP, 2018-03-18:
So let me know if there's anything else you need.

gavlig, 2018-06-11:
Hi guys, any updates on this issue? If you need any help, I have a trinity here as well.

ahoneybun, 2018-06-16:
I also have a Naga Trinity if I can help any to get this working.

d3is, 2018-08-01:
Was anyone working on this still or have any progress complete? I have a Trinity and would love to get this thing supported. I am unfamiliar with the codebase but willing to try and learn and get this thing supported. Or I can contribute any logs, data, or testing.

GetchaDEAGLE, 2018-08-05:
I just made the jump to Linux for my main OS and ordered this mouse. I'd be more than happy to help support this effort as well in any way I can.

Also, I just wanted to say thanks to the developers of this great project. I'm beyond impressed as out of all the options for mice on Linux, this project alone has made the choice of using Razer rather easy. Everything else is frail in comparison.

z3ntu, 2018-09-11:
@d3is (or someone else who's interested) please message me on IRC (z3ntu on freenode), Matrix (@z3ntu:matrix.org) or Telegram (@z3ntu) to try test something out.

quimbydogg, 2018-12-08:
I recently picked up one of these mice and was glad to see this project.  I downloaded the naga trinity branch of openrazer referenced earlier (at commit 59a89e7b89e0a2b25534b0feff45ce8758f5f9ba).  I'm able to change the brightness settings using the command mentioned earlier but polychromatic controller gives me a message that the device is not recognized by the daemon.  I suspect that is expected behavior at this point.  Please let me know how I can help support/testing for this device.  Would be great to have the ability to change colors.

presianbg, 2018-12-13:
Hi, 
I just bought Razer Naga Trinity and can help with whatever you need to make it work under Linux :+1:

FeralBytes, 2018-12-15:
@z3ntu what do you need next?

imperative, 2018-12-16:
Do extra mouse buttons work as mouse buttons already? (Without any drivers they are shown in X server as keyboard events for keys "1", "2", "3", etc - makes it impossible to bind them to anything without sacrificing the actual keyboard keys. )

Cyriou, 2018-12-25:
Hello,
I've a trinity razer and I was wondering how is the project ? And if I can help
Tanks :-)

FullerBot, 2019-02-02:
Just picked up a Razer Naga Trinity
More than happy to do some testing if you'd like- I've got limited command-line skills, but I'm more than happy to try whatever you guys would like.

damianatorrpm, 2019-02-04:
Reading the diff between the trinity branch and the original it seems that only the device id's are added. Wouldn't it be better to merge this (as it doesn't interfere with other code) and label it experimental instead of not merging it? @z3ntu

cdm1322, 2019-02-09:
Hello,
I'm interested in helping along further support of the trinity mouse. I have one personally and could possibly help with development as well.

Let me know if I can help.

justinnichols, 2019-04-05:
I attempted to re-apply @z3ntu initial commits in a branch off the latest master and it indeed allowed the Trinity to be recognized, however, brightness didn't work.  If I set it to use get_brightness and set_brightness and change his code such that it does a standard led_brightness instead of matrix_brightness, the brightness will change on the mouse (for the thumbgrid, logo, and scroll wheel, as a single setting), but the setting in polychromatic won't stick (keeps bouncing to zero).

In any case, I took my best shot at working some on this and didn't get very far (driver software and reverse engineering binary specs is not in my wheelhouse).  I was wondering if there has been any other progress on this.

hahnt9009, 2019-04-08:
Hi everyone, I was also curious on progress of this support. I am also willing to do whatever I can to help, don't have a lot of experience with this sort of thing but would love to have the Trinity working to be able to adjust lighting and/or dpi and whatnot on linux, I have a Trinity and Antergos / Win10, let me know if I can be of assistance!

Maltahl, 2019-04-10:
@justinnichols Could you change settings despite the lightning not sticking ?

justinnichols, 2019-04-10:
> @justinnichols Could you change settings despite the lightning not sticking ?

The brightness setting on the mouse did indeed change, but the setting in polychromatic simply didn't stay where the setting actually is.  I would drag to the right, the brightness would increase on the mouse once I completed the drag, but the setting in polychromatic would move the slider all the way back to the left.

Maltahl, 2019-04-10:
@justinnichols i tried getting it to appear in polychromatic by doing make install as explained further up with feature_naga_trinity_0x0067 branch but with no luck.

What steps did you take to get it to work for you ?

justinnichols, 2019-04-10:
> @justinnichols i tried getting it to appear in polychromatic by doing make install as explained further up with feature_naga_trinity_0x0067 branch but with no luck.
> 
> What steps did you take to get it to work for you ?

@Maltahl I'll try to set it up again as soon as possible and then will commit the code to a branch in my fork.  I'll then reply here with the steps to build/load the new driver.  It'll be later today, probably, before I can get to it, though.

justinnichols, 2019-04-10:
@Maltahl I was able to get a few minutes to work this again.  It actually didn't require changing the matrix_brightness code as I had originally thought, it only required updating the mouse.py to include the Naga Trinity.

To see the changes I made beyond what @z3ntu had already done, view the following commit:
https://github.com/justinnichols/openrazer/commit/63d4a80ce4722580762749a7857edfd458a8b6e8

In order to get the driver to work after making changes, you need to do the following:

1. Run: `sudo gpasswd -a yourusernamehere plugdev`  (this adds your user to the plugdev group)
2. Run the commands below to build/install the driver, but after the first time, reboot the computer to make sure the razermouse kernel module gets loaded (modprobe didn't really work for me for some reason).
3. Verify razermouse is loaded using: `lsmod |grep -i razermouse`

I'm using Pop!_OS 18.10 (based on Ubuntu 18.10) and this is the script I would run in the root of the project (I created a shell script for it), to update the driver:

```
#!/bin/sh
make clean
make all
./scripts/build_debs.sh
sudo dpkg -i ./dist/*.deb
```

I then have to restart the daemon in polychromatic to see any changes.

Once your mouse is recognized and the settings show up (though it'll have the photos for the Naga Chroma), when you drag and release the brightness control, you'll see the mouse brightness did indeed change on your hardware (but the setting doesn't stick).

I'm fairly sure this path isn't exactly on the right track.  I'm expecting that each LED has independent brightness (thumb grid, scrollwheel, and logo), and since the Trinity has multiple thumb attachments, the driver will need to handle all three for all the various settings.  I could never get the independent brightness controls to work, only the global set_brightness worked, somewhat.

Maltahl, 2019-04-10:
@justinnichols no dice with getting it to show in polychromatic after cloning your updated git and following your instructions. i also tried a reboot.

justinnichols, 2019-04-10:
@Maltahl I don't believe it will show if the output of `lsmod |grep -i razermouse` isn't something to the effect of:
```
razermouse             65536  0
hid                   126976  3 usbhid,hid_generic,razermouse
```

You should check the daemon log to see if the Trinity shows up there as well.  The log is located here: `~/.local/share/openrazer/logs/razer.log`.

The log will show something like:
```
2019-04-10 13:24:21 | razer                          | DEBUG    | Loaded device specification: RazerNagaTrinity ------------------------ (1532:0067)
```
and
```
2019-04-10 13:24:21 | razer.device0                  | INFO     | Initialising device.0 RazerNagaTrinity
2
```

Beyond that, I'm not sure I'll be much help to you because there would be something different in your setup that is affecting this.

Maltahl, 2019-04-10:
@justinnichols wierd it does not load the RazerNagaTrinity device spec when starting after looking in ~/.local/share/openrazer/logs/razer.log .
ubuntu 18.04 if that has anything to say.

justinnichols, 2019-04-10:
> @justinnichols wierd it does not load the RazerNagaTrinity device spec when starting after looking in ~/.local/share/openrazer/logs/razer.log .
> ubuntu 18.04 if that has anything to say.

@Maltahl I'm sorry, I don't think I'll be much more help.  Just make sure you also added your user to that `plugdev` group.  If you did everything I listed, then that's all I did to make it work.  Keep in mind that if you cloned my fork, you have to checkout the `feature_naga_trinity_0x0067_updated` branch before building/installing.

Maltahl, 2019-04-10:
@justinnichols can i have you try compile the .deb files for me ? it still wont add feature_naga_trinity_0x0067_updated in the git ( or it says it does but it really does not)

justinnichols, 2019-04-10:
@Maltahl I do not know if it will work on Ubuntu 18.04, but here are the .deb files (as a zip file you'll have to extract).

[openrazer-driver-files-justinnichols-2019-04-10-13-23.zip](https://github.com/openrazer/openrazer/files/3065231/openrazer-driver-files-justinnichols-2019-04-10-13-23.zip)

[15 more replies]

commit 6e7be543 (2019-10-23): Add support for Razer Naga Trinity

Fixes #541
closed 2019-10-23 by z3ntu