openrazer/openrazer #542: Linux Mint 18.3 DeathAdder Elite "Device not recognized" on polychromatic

escobera, 2018-03-02:
Hello there,
I've installed this tool and polychromatic from the stable ppas but can't get my mouse to work under it.

`sudo dkms install openrazer-driver/1.0.0; sudo modprobe razerkbd
python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)` gives

`Module openrazer-driver/1.0.0 already installed on kernel 4.13.0-36-generic/x86_64
[]
`

z3ntu, 2018-03-02:
`sudo modprobe razerkbd `?

escobera, 2018-03-02:
> sudo modprobe razerkbd?
No message shows up

z3ntu, 2018-03-02:
Then `lsmod | grep razer` ?
Seems like the components the kernel driver, daemon and python lib work fine...

escobera, 2018-03-02:
```
$ lsmod | grep razer
razerkbd               57344  0
razermouse             65536  0
hid                   118784  4 razermouse,hid_generic,usbhid,razerkbd
```

z3ntu, 2018-03-02:
`lsusb | grep 1532` ?

escobera, 2018-03-02:
```
$ lsusb | grep 1532
Bus 003 Device 003: ID 1532:005c Razer USA, Ltd 
```

maxitenerife, 2018-03-02:
I'm experiencing he same problem on Ubuntu 16.04. :(

escobera, 2018-03-02:
Yeah, I forgot to mention I'm on Linux Mint 18.3

z3ntu, 2018-03-02:
@maxitenerife Also the same device?

@escobera Output of `ls -al /sys/bus/hid/drivers/razermouse/0*/` ?

escobera, 2018-03-02:
```
$ ls -al /sys/bus/hid/drivers/razermouse/0*/
ls: cannot access '/sys/bus/hid/drivers/razermouse/0*/': No such file or directory
```

maxitenerife, 2018-03-03:
ls -al /sys/bus/hid/drivers/razermouse/0*/
Bus 002 Device 005: ID 1532:004d Razer USA, Ltd 

ls -al /sys/bus/hid/drivers/razermouse/0*/
ls: cannot access '/sys/bus/hid/drivers/razermouse/0*/': No such file or directory

No message shows upwith sudo modprobe razerkbd, same with sudo modprobe razermouse

It seems as the same problem that others are experiencing under Mint.

escobera, 2018-03-06:
@z3ntu anything else I can do to debug this issue?

z3ntu, 2018-03-06:
Is there something with `sudo journalctl -b | grep -i razer_mount` ?
If there is no error I guess udev is not getting triggered correctly.

escobera, 2018-03-07:
@z3ntu 
`sudo journalctl -b | grep -i razer_mount` shows nothing

z3ntu, 2018-03-07:
Then it really seems like udev is being weird. Can you try installing an older version of the driver to see if the issue was introduced somewhere?

escobera, 2018-03-07:
I can do that, but I don't know how to. The ppa seems to carry only the latest version.

z3ntu, 2018-03-07:
Clone the repo, check out an old version (maybe v2.2.0 if that version supports this device), then look at https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint how to build and install the package

z3ntu, 2018-03-08:
Okay so this issue is Linux Mint 18.3 (based on Ubuntu 16.04) and Ubuntu 16.04... I have downloaded the `openrazer-kernel-modules-dkms_2.2.2~ubuntu16.04.1_amd64.deb` file from the ppa and the udev rules file is in there... And master is literally v2.2.2 ...

z3ntu, 2018-05-22:
No response - closing.

closed 2018-05-22 by z3ntu