openrazer/openrazer #543: Hyperflux Support

mrmakeit, 2018-03-02:
The Razer Mamba Hyperflux and Firefly Hyperflux seem to be considered a single device.

Mamba:
Product ID: RZ83-02480100
Model Number: RZ01-0248

Firefly:
Product ID: RZ83-02480100
Model Number: RZ02-0248

Firefly plugged in:
[lsusb -d 1532:0068 -v](https://pastebin.com/iBr7UF4J)
[ls -lah /dev/input/by-id/](https://pastebin.com/E3xP3PCY)

Mamba plugged in:
[lsusb -d 1532:0069 -v](https://pastebin.com/25qG6nb8)
[ls -lah /dev/input/by-id/](https://pastebin.com/TF7Zj0C5)

I have to repair my Windows VM, then I'll get pcap's and screenshots.

pashtet04, 2018-09-02:
Could you add support of this device please?

redaxmedia, 2019-02-08:
### Mamba

[image]
[image]
[image]
[image]
[image]
[image]

### Firefly

(most of the settings are disabled)

[image]
[image]

redaxmedia, 2019-02-09:
# Mamba Hyperflux
## Settings
[image]
[image]
[image]
## Performance
[image]
## Lightning
[image]
## Calibration
[image]
# Mouse Mat
## Lightning
[image]

hedmo, 2019-12-13:
@z3ntu what does milestone mean? i am on the hyperflux road to.i am running polychromatic-0.4.0 and seeing an unrecognized:1532:0068 in the menu.

regards hedmo

z3ntu, 2019-12-14:
@hedmo Ignore the milestone, I use it more or less as a label for me to know what generation of Razer devices this is from

hedmo, 2019-12-14:
@z3ntu .thanks for the reply .but how about the hyperflux ? is it going to  be supported at any time.i dont understand if it is or going to be supported..

regards

z3ntu, 2019-12-15:
I'm lately busy with other projects so I don't have much time for OpenRazer. This device should be compared to other devices easy to implement, you could take a look at recent merged pull requests for a reference on what has to be done for the device to be supported.

hedmo, 2019-12-15:
@z3ntu okay thanks.i understand. i am going try to help an add this .as i understand this is like the firefly and just need to change the id and stuff to make it work.maybe the hero @nightsky30 can help me this time to?.but for now i am going to find a pull requests for an added device

regards hedmo

nightsky30, 2019-12-15:
@hedmo Looks like the Mamba Hyperflux could be similar to the Lancehead as wired/wireless options but without the side LEDs. You could use #887 as an example to work from.

hedmo, 2019-12-15:
@nightsky30 .at the moment i am trying to use the firefly setup by patching : 
mouse_mat.py
razermousemat_driver.h
and
99-razer.rules
with the 0068 id.the device is recognized but it is not changing the rgbs so i think there is more files i need to patched.i will check #887 and i am going to see if i can find the commit that adds the firefly and work from there.

rerards

hedmo, 2019-12-16:
@nightsky30 i have manage to use the Goliathus setup to work with firefly_hyperflux. the firefly setup did not work and i dont know why.i have edit the files to get the new product working at [https://github.com/hedmo/openrazer](url) .the problem now is that Goliathus and firefly are configured different and i dont know how to add wave to firefly_hyperflux
[image]

regards hedmo

nightsky30, 2019-12-16:
A couple things.  Not sure if the _Goliathus Chroma_ or _Goliathus Chroma Extended_ support wave through software or hardware, but from a video on Razer's site they appear to support the wave effect _somehow_.  BUT, regardless wave was not add for those devices in openrazer.  The Goliathus devices use the extended matrix effects and ZERO_LED vs the regular firefly that is using the standard matrix effects and BACKLIGHT_LED.

You can try adding the following and see if the firefly hyperflux supports the wave effect through hardware.

Add 'set_wave_effect' in the list of methods for the firefly hyperflux class in your version of mouse_mat.py.

Then I would try modifying razer_attr_write_mode_wave in razermousemat_driver.c so that it includes a switch statement to account for the different device types with a case statement for the firefly hyperflux, and a default statement for the normal firefly.  The firefly hyperflux appears to be using extended matrix effects with ZERO_LED.
```
    ...
    struct razer_report report = {0};

    switch (usb_dev->descriptor.idProduct) {
    case USB_DEVICE_ID_RAZER_FIREFLY_HYPERFLUX:        
        report = razer_chroma_extended_matrix_effect_wave(VARSTORE, ZERO_LED, direction);
        break;

    default:
        report = razer_chroma_standard_matrix_effect_wave(VARSTORE, BACKLIGHT_LED, direction);
        break;
    }

    razer_send_payload(usb_dev, &report);

    return count;
    ...
```

hedmo, 2019-12-16:
@nightsky30 thanks for the hint but no go but now i can turn the light on and of ... i have set MATRIX_DIMS = [1, 15] in mouse_mat.py  but only one (the first) is working and lights all.

hedmo, 2019-12-26:
@nightsky30 @z3ntu  i am stuck. i dont know where to go from here with the wave function.i will try to give you as much info as i can :

what works = spectrum,breath,static and brightness
what does now work= wave and in RazerGenie the matrix function= set color per zone 1-15(only zone 1 works and lights all).
the  MATRIX_DIMS = [1, 15] is the light zones as i understand? and that one should be set to MATRIX_DIMS = [1, 1] as i would like to have it but if i put it like that,the lights get fuzzy and only zone 1 will light up fully.

this is as far i have manage to get.it would be nice fix the last lose end (wave) to get this pulled
for other users to control the hyperflux and for me to move to the hyperflux mouse and the nari ultimate.i understand if time is a problem.

regards hedmo

z3ntu, 2019-12-26:
As for effects, Synapse only uses the 'static' effects in hardware it seems, so I can't help you with wave.

For matrix dimensions, Synapse only uses row 0 column 0 start (& column 1 end), so that would match the `MATRIX_DIMS = [1, 1]`

hedmo, 2019-12-26:
@z3ntu. okey as i understand,this is as good as it gets but i will try MATRIX_DIMS = [1, 1] one more time .then i like to pull my https://github.com/hedmo/openrazer to yours but i dont know how?

regards

nightsky30, 2019-12-26:
@hedmo It seems like you cloned openrazer and pushed rather than forked from the official openrazer.  So **I think** the upstream connection is missing.  You could try adding the official openrazer repo as an upstream to your local master.  Then fetch --all and merge on your local master with the remote upstream.  And I think finally push to your openrazer remote repo before creating a pull request.

Otherwise, fork the official openrazer, clone your fork to a separate folder, reapply your changes to your forked local master, then push to your remote fork, and create a pull request.

Forgive me if I'm not clear enough or I had the wrong impression ^^

hedmo, 2019-12-27:
@nightsky30 

yes i had a clone. this is new to me. i had to clone mine,delete my repo,fork openrazer,readd my clone to the repo. my clone was older then openrazer so i had to manually update the razer orbweaver chrome update.hoping to get help with my pull request #981

thank you for all the help

regards hedmo

z3ntu, 2020-05-02:
Fixed with #981

closed 2020-05-02 by z3ntu

Nama, 2022-08-06:
I can't set the DPI of the Mamba Hyperflux. Shouldn't it be added with this? Only Firefly is showing up (which is prolly expected), but there aren't any DPI or button settings.