openrazer/openrazer #545: Toggle multimedia keys without fn?

ebertn, 2018-03-06:
I'm sorry if this is the wrong place for questions. If so please point me in the right direction. 

Does open razer support toggling fn in order to use multimedia keys? This is something I know razer synapse does that I would like to do now that I have moved over to ubuntu.

z3ntu, 2018-03-06:
Not that I know. But it seems like more of a BIOS/UEFI setting (at least on Blade laptops).

stealthybox, 2019-04-24:
@z3ntu a BIOS would have made sense, but there is none on the Razer Blade 15 Advanced.

Razer Synapse is where this is configured on Windows, so it should be possible to accomplish this in the driver.

z3ntu, 2019-04-26:
It probably would make sense to be put into the driver but as the amount of people wanting that feature seems to be pretty low :) it's not a high priority. Contributions welcome of course!

vevake, 2019-05-18:
@z3ntu On doing the following command I can toggle the fn key.
1. Navigate to /sys/bus/hid/drivers/razerkbd/[device id]
2. echo -n "1" > fn_toggle

But every time I restart the machine it toggles back again, so I have to redo this on startup.... Is there a permanent solution to doing this.

Thanks.

stealthybox, 2019-05-20:
@vevake, what keyboard are you using?
That doesn't work for any of the device id's on my Razer Blade 15 (2019)
```bash
for kbd in $(find /sys/bus/hid/drivers/razerkbd/ -name '0003*'); do
  (cd $kbd; echo $kbd; echo -n 1 | sudo tee fn_toggle);
done
/sys/bus/hid/drivers/razerkbd/0003:1532:023A.0005
tee: fn_toggle: Permission denied
1/sys/bus/hid/drivers/razerkbd/0003:1532:023A.0003
tee: fn_toggle: Permission denied
1/sys/bus/hid/drivers/razerkbd/0003:1532:023A.0004
tee: fn_toggle: Permission denied
1
```

vevake, 2019-05-20:
@stealthybox I use Razer blade stealth late 2017.

I had the same issue before. You have to be under root permissions to write to fn_toggle.

1. do "sudo su"
2. Navigate to /sys/bus/hid/drivers/razerkbd/[device id]
3. echo -n "1" > fn_toggle

This should work.

stealthybox, 2019-05-20:
@vevake my loop is already using `echo -n 1 | sudo tee fn_toggle` to write the file.
That file does not exist on my system, and I cannot create it as root.

vevake, 2019-05-20:
@stealthybox The file did not exist in my system as well, but I was able to create it using the above commands in terminal.

vevake, 2019-05-26:
I was able to toggle fn key permanently by using the commands in /etc/rc.local file, so that the toggle function is done automatically on system startup.

stealthybox, 2019-06-10:
@vevake 
Yeah, I just double checked, but using root on my system in a variety of ways doesn't allow me to write to that file:
```bash
root@linuxblade:/sys/bus/hid/drivers/razerkbd/0003:1532:023A.0003# echo -n "1" > fn_toggle
-bash: fn_toggle: Permission denied

root@linuxblade:/sys/bus/hid/drivers/razerkbd/0003:1532:023A.0004# echo -n "1" > fn_toggle
-bash: fn_toggle: Permission denied

root@linuxblade:/sys/bus/hid/drivers/razerkbd/0003:1532:023A.0002# echo -n "1" > fn_toggle
-bash: fn_toggle: Permission denied
```
Thanks for the pointer though!
Perhaps my laptop's kbd driver has a different name.
How did you find out the `fn_toggle` file name? Did you grep some code or docs somewhere?

vevake, 2019-06-10:
@stealthybox 
I got the pointer from [issue](https://github.com/openrazer/openrazer/issues/186) .. I tried it and it worked for me...

May be the problem for you is the file permissions....if you sort the permissions it should work

stealthybox, 2019-06-24:
> May be the problem for you is the file permissions....if you sort the permissions it should work

This is not the problem for me -- I'm root. It's being denied for another reason.
Thanks for the pointer :+1:

mike-lloyd03, 2024-07-21:
I'm way late to the party here, but the `fn_toggle` file doesn't exist for me on my Razer Blade Advanced 2020. And I cannot create it because you cannot create files in `sysfs`.

Is there another way to enable this on my laptop?