openrazer/openrazer #546: razer blade stealth late 2016 issue with restart and power off

dominikS007, 2018-03-08:
When turning off the system it will never turn off completely it will crash while trying to turn off, the same goes for the restart. The system that I have is:

Razer Blade Stealth late 2016 UHD 
CPU: Intel® Core™ i7-7500U CPU @ 2.70GHz × 4 
GPU: Intel® HD Graphics 620 (Kaby Lake GT2)
Gnome: 3.26.2

z3ntu, 2018-03-08:
Does the same thing happen without the drivers installed?

dominikS007, 2018-03-08:
I am a new user of Ubuntu and do not know many commands, which command would I use to uninstall the whole driver ?

z3ntu, 2018-03-08:
`sudo apt purge openrazer-meta` and maybe then `sudo apt autoremove`

dominikS007, 2018-03-08:
I have just restarted the system and the system did not turn off but crashed while doing it.

z3ntu, 2018-03-08:
If it's still happening without driver installed (check with `dpkg -l | grep openrazer`) then it's not the fault of this software.

closed 2018-03-08 by z3ntu