openrazer/openrazer #549: 'RazerBlackWidowChromaTournamentEdition' object has no attribute 'setMacroEffect'

DanEble, 2018-03-16:
```
2018-03-15 20:34:14 | razer.device0.keymanager       | INFO     | Got macro combo
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 144, in run
    self._poll_epoll(poll_object, event_file_map)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 174, in _poll_epoll
    self._parent.key_action(date, key_code, key_action)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/misc/key_event_management.py", line 433, in key_action
    self._parent.setMacroEffect(0x01)
AttributeError: 'RazerBlackWidowChromaTournamentEdition' object has no attribute 'setMacroEffect'
```

On Ubuntu 17.10, using the "stable" PPA, openrazer-daemon is hitting this error when I press the macro-record key (fn-F9).  After this error, it no longer responds to the game mode or backlight keys.  Attached is the [full output](https://github.com/openrazer/openrazer/files/1817563/openrazer-daemon-output.txt) of running the daemon in the foreground, pressing the macro-record key, and then interrupting the daemon.

My keyboard is labeled as follows.

> BLACKWIDOW TE CHROMA V2
> MODEL NO. RZ03-0219
> PRODUCT NO. RZ03-02190200-R3U1

The firmware version is v1.3.

I can confirm that the macro_led_state and macro_led_effect files documented on the [wiki](https://github.com/openrazer/openrazer/wiki/Using-the-keyboard-driver) are operative.  With the backlight on, echoing 1 to macro_led_state shuts off the light under the F9 key; then echoing 1 to macro_led_effect causes the light under the F9 key to flash.  Note that neither of these effects is noticeable when the backlight is off.

z3ntu, 2018-03-30:
It seems that the code assumes that all keyboards has the M keys

DanEble, 2018-07-28:
Until now, I thought that a robot being a jerk is something that only happens in cartoons.  Age doesn't make a bug less worthy of fixing.

Regards to my fellow humans,
Dan

kanarok, 2018-07-31:
I have that keyboard and hacked some stuff together to have actual M Keys on it. Will look into this bug as soon as I can spare some time. I want it to work, too.