openrazer/openrazer #552: Deamon problem

starcraft2fan, 2018-03-20:
Hi, i have Razer Deathadder and Razer Kraken v2. I installed Polychromatic Controller but he's dont work. RazerGenie too. I have this code when i'm trying to open these applications: 
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-03-20 02:45:43 CET; 25s ago
     Docs: man:openrazer-daemon(8)
  Process: 2392 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 2392 (code=exited, status=1/FAILURE)

mar 20 02:45:43 bartek-pc systemd[664]: Starting Daemon to manage razer devices in userspace...
mar 20 02:45:43 bartek-pc openrazer-daemon[2392]: Starting daemon.
mar 20 02:45:43 bartek-pc openrazer-daemon[2392]: 2018-03-20 02:45:43 | razer                          | CRITICAL | User is not a member of the plugdev group
mar 20 02:45:43 bartek-pc openrazer-daemon[2392]: Stopping daemon.
mar 20 02:45:43 bartek-pc systemd[664]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
mar 20 02:45:43 bartek-pc systemd[664]: openrazer-daemon.service: Failed with result 'exit-code'.
mar 20 02:45:43 bartek-pc systemd[664]: Failed to start Daemon to manage razer devices in userspace.

Any help? Thanks :)

z3ntu, 2018-03-20:
Please read the log

> mar 20 02:45:43 bartek-pc openrazer-daemon[2392]: 2018-03-20 02:45:43 | razer | CRITICAL | User is not a member of the plugdev group

neptunerx, 2018-03-20:
Same issue with my user in the plugdev group 

$ id winter
uid=1000(winter) gid=1000(winter) groups=1000(winter),0(root),4(adm),8(mail),27(sudo),46(plugdev),129(docker)

z3ntu, 2018-03-21:
Have you rebooted after adding yourself to the group?

raetiacorvus, 2018-04-20:
In my case I still had the error message in my journal even tough the daemon started fine for the user I logged in with and which was in plugdev.

Figured out that you also need to add your login manager user (gdm in my case) to plugdev.

closed 2018-04-20 by z3ntu