openrazer/openrazer #553: Headset Tiamat 7.1 v2

alcos0, 2018-03-22:
Hi!
Could you guys please support the headset Tiamat 7.1 v2? I have one and I am willing to build and test your drivers.
Thanks!

z3ntu, 2018-03-22:
Yeah sure. 
You can join our community chat, either via Telegram (https://t.me/joinchat/A1E4RT9TxhnxyetdQCSj7Q) or via Matrix (https://matrix.to/#/#openrazer:matrix.org), that way we can help you much easier.

alcos0, 2018-03-22:
By the way, here is the data I was suppose to include when opening the issue:

```
$ lsusb -d 1532:0f03 -v
Bus 001 Device 011: ID 1532:0f03 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f03 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Tiamat 7.1 V2
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      18
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```
$ ls -lah /dev/input/by-id/ | grep Tiamat
lrwxrwxrwx 1 root root   9 mar 22 11:48 usb-Razer_Razer_Tiamat_7.1_V2-if01-event-kbd -> ../event8
```
Later today I'll capture the packets needed.

alcos0, 2018-03-23:
The available effects:
[image]

Each packet file has a pretty straight forward name including the effects I was changing in between and the selected color(s).
[packets.tar.gz](https://github.com/openrazer/openrazer/files/1839684/packets.tar.gz)

jim0510, 2019-02-08:
I recently brought the headset as well. Is there anything that I can contribute?

CalcProgrammer1, 2020-04-09:
I got one of these recently and am going to look into adding support for it.

JustSomeGeek, 2023-05-12:
Just adding a "Me too" for wanting support added. Happy to test anything on Arch, if I can.

z3ntu, 2023-05-12:
@JustSomeGeek Feel free to contribute to PR #1091, someone with the device needs to finish that one up.

JustSomeGeek, 2023-05-12:
> @JustSomeGeek Feel free to contribute to PR #1091, someone with the device needs to finish that one up.

Have  commented on it, thanks.