openrazer/openrazer #554: Daemon is not responding Fedora 27

kentgr, 2018-03-23:
I got a problem since a few weeks back that the daemon is not responding and read the case https://github.com/openrazer/openrazer/issues/461 but it does not help. 

This is the error i get: 
python3 -c "from openrazer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/openrazer/client/__init__.py", line 38, in __init__
    device = _RazerDeviceFactory.get_device(serial)
  File "/usr/lib/python3.6/site-packages/openrazer/client/device.py", line 51, in get_device
    device_type = device_dbus.getDeviceType()
  File "/usr/lib64/python3.6/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib64/python3.6/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Traceback (most recent call last):
  File "/usr/lib64/python3.6/site-packages/dbus/service.py", line 654, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib64/python3.6/site-packages/dbus/service.py", line 246, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: getDeviceType is not a valid method of interface razer.device.misc

kentgr, 2018-03-23:
pkill -e -f openrazer-daemon; openrazer-daemon -Fv 
Starting daemon.
2018-03-23 16:28:03 | razer                          | INFO     | Initialising Daemon (v2.2.2). Pid: 10088
2018-03-23 16:28:03 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2018-03-23 16:28:03 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
[... cut ...]
2018-03-23 16:28:04 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
2018-03-23 16:28:04 | razer                          | INFO     | Serving DBus
^C2018-03-23 16:28:31 | razer                          | INFO     | Stopping daemon on signal 2
2018-03-23 16:28:31 | razer.device2.keymanager       | DEBUG    | Stopping key manager
2018-03-23 16:28:31 | razer.device3.keymanager       | DEBUG    | Stopping key manager
2018-03-23 16:28:31 | razer.device3.ripplemanager    | DEBUG    | Closing Ripple Manager
^CStopping daemon.

z3ntu, 2018-03-23:
You can apply the diff of f5c4fe6 manually to the file `/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/keyboards.py` or wait for the next release (which might take some time though).

commit f5c4fe6d (2018-03-23): Re-add get_device_type_keypad for nostromo I accidentally removed

Fixes efca0087a2a9 ("Simplify common daemon methods")
Fixes #554
closed 2018-03-23 by z3ntu