openrazer/openrazer #556: Daemon not running/giving error

Philenst, 2018-03-24:
If I try to start the daemon I just get

> Failed to start openrazer-daemon.service: Unit openrazer-daemon.service not found.

But If I run openrazer-daemon -Fv I get

`Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 178, in <module>
    run()
  File "/usr/bin/openrazer-daemon", line 173, in run
    logger=logger)
TypeError: __init__() got an unexpected keyword argument 'chdir'
phil@box:/$ `

could anyone help me out?

z3ntu, 2018-03-24:
Which distro (and version) do you use? This is caused due to an old version of python3-daemonize but that should be installed automatically with the packages.

Philenst, 2018-03-24:
Distributor ID:	elementary
Description:	elementary OS 0.4.1 Loki
Release:	0.4.1
Codename:	loki

It's a distro more for the looks than anything else, not sure why it doesn't install with the package?

z3ntu, 2018-03-24:
Which commands did you run to install the drivers?

lah7, 2018-03-24:
It certainly works on elementaryOS 0.4.1, did you install via the PPA or manually via packages?

If it's the latter, you can [grab a Debian package of the newer `python3-daemonize`](https://launchpad.net/~openrazer/+archive/ubuntu/stable/+files/python3-daemonize_2.4.7-xenial_all.deb) (2.4.7) but it would be better to add the PPA to receive future updates.

    sudo add-apt-repository ppa:openrazer/stable
    sudo apt update

Philenst, 2018-03-24:
I just used the Debian instructions that you guys provide, I'm pretty sure that I used the PPA to install it but I can try it again.

Philenst, 2018-03-24:
After reinstalling and purging everything, ran a clean install from the PPA and it seems to work, thanks guys

closed 2018-03-24 by Philenst