openrazer/openrazer #558: Daemon fails to start for DeathStalkerChroma with "UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb6 in position 2: invalid start byte"

monovertex, 2018-03-25:
Full output of `openrazer-daemon -Fv` as follows:

```
Starting daemon.
2018-03-25 18:28:24 | razer                          | INFO     | Initialising Daemon (v2.2.2). Pid: 6168
2018-03-25 18:28:24 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerKrakenClassic ---------------------- (1532:0501)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerMambaTE ---------------------------- (1532:0046)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerNagaChroma ------------------------- (1532:0053)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaverChroma -------------------- (1532:0207)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2011 ------------------------- (1532:0013)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2013 ------------------------- (1532:0039)
2018-03-25 18:28:24 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
[... cut ...]
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb6 in position 2: invalid start byte

Exception ignored in: <object repr() failed>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 426, in __del__
    self.close()
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 339, in close
    if not self._is_closed:
AttributeError: 'RazerDeathStalkerChroma' object has no attribute '_is_closed'
Stopping daemon.
```

Let me know if I can provide any other info.

z3ntu, 2018-03-25:
Can you run `tail /sys/bus/hid/drivers/razermouse/0*/device_serial`?

And the only way to easily ignore a device is by removing it from the udev rules at (`/usr`)`/lib/udev/rules.d/99-razer.rules`.

monovertex, 2018-04-11:
Hi, sorry, I've been busy the past couple of weeks and haven't used that machine anymore. I ran the command, here's the output:

```bash
 ~  tail /sys/bus/hid/drivers/razermouse/0*/device_serial
PM002117823462
```

Regarding `99-razer.rules`, could I just remove keyboard id `0204` from this line to fix my issue?
```bash
ATTRS{idProduct}=="010d|010e|0111|0113|011b|011a|010f|0201|0202|0203|0204|0205|0207|0208|0209|020f|0210|0211|0214|0216|021a|0217|021a|021e|021f|0220|0221|0224|0225|022d|022f|0232"
```

commit cf4c2bf4 (2018-05-20): Handle UnicodeDecodeError in get_serial

which can get thrown due to malformed serials

Fixes #594 and fixes #558
commit af84bd6c (2018-05-20): Handle UnicodeDecodeError in get_serial

which can get thrown due to malformed serials

Fixes #594 and fixes #558
closed 2018-05-20 by z3ntu