openrazer/openrazer #563: Can't modprobe razerkbd

dannyss6, 2018-04-01:
Yeah so I'm not really experienced with drivers and such. I installed openrazer and polychromatic and poly tells me that the daemon is running and the mouse (naga 2014) is detected. It tells me to use modprobe to bind the mouse. I tried using modprobe razerkbd but the file isn't in the folder /lib/modules/Kali
Thanks for helping me solve this.

z3ntu, 2018-04-01:
What's the output of `sudo dkms install openrazer-driver/1.0.0` ?

dannyss6, 2018-04-01:
Thanks looks like i didn't had headers installed. Also is there a 'made to' way to make it boot with --as-root flag as default so I don't have to manually do it ?

dannyss6, 2018-04-01:
Oops i just saw the troubleshooting page lol

z3ntu, 2018-04-01:
Why are you running the daemon as root...?

dannyss6, 2018-04-01:
Most of the things i do needs root so i just login as root no big deal

z3ntu, 2018-04-01:
Well, your choice. You could edit the systemd unit file and add the parameter there if you really have to. (filename will be printed with `systemctl --user status openrazer-daemon.service`). But your issues seem to be solved, so I'm closing this.

closed 2018-04-01 by z3ntu