openrazer/openrazer #564: Keyboard only show white light on Razer Blade Stealth (Late 2017)

ykyuen, 2018-04-03:
Hi all,

I tried both polychromatic and razercommander but still not able to have keyboard lights with different colors.

The openrazer-daemon is running and did receive the command from razercommander.
```
$ systemctl --user status openrazer-daemon.service
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-04-03 17:24:26 HKT; 10min ago
     Docs: man:openrazer-daemon(8)
 Main PID: 2040 (openrazer-daemo)
   CGroup: /user.slice/user-1000.slice/user@1000.service/openrazer-daemon.service
           └─2040 openrazer-daemon

Apr 03 17:34:07 camus openrazer-daemon[2040]: 2018-04-03 17:34:07 | razer.device0                  | DEBUG    | DBus call set_static_effect
Apr 03 17:34:07 camus openrazer-daemon[2040]: 2018-04-03 17:34:07 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1733A12800011, 'setStatic', dbus.Byte(255), dbus.Byte(255), dbus.Byte>
Apr 03 17:34:30 camus openrazer-daemon[2040]: 2018-04-03 17:34:30 | razer.device0                  | DEBUG    | DBus call set_brightness
Apr 03 17:34:30 camus openrazer-daemon[2040]: 2018-04-03 17:34:30 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1733A12800011, 'setBrightness', 66)
Apr 03 17:34:30 camus openrazer-daemon[2040]: 2018-04-03 17:34:30 | razer.device0                  | DEBUG    | DBus call set_static_effect
Apr 03 17:34:30 camus openrazer-daemon[2040]: 2018-04-03 17:34:30 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1733A12800011, 'setStatic', dbus.Byte(204), dbus.Byte(0), dbus.Byte(0>
Apr 03 17:34:49 camus openrazer-daemon[2040]: 2018-04-03 17:34:49 | razer.device0                  | DEBUG    | DBus call set_brightness
Apr 03 17:34:49 camus openrazer-daemon[2040]: 2018-04-03 17:34:49 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1733A12800011, 'setBrightness', 229)
Apr 03 17:34:49 camus openrazer-daemon[2040]: 2018-04-03 17:34:49 | razer.device0                  | DEBUG    | DBus call set_static_effect
Apr 03 17:34:49 camus openrazer-daemon[2040]: 2018-04-03 17:34:49 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBladeStealthMid2017:BY1733A12800011, 'setStatic', dbus.Byte(204), dbus.Byte(0), dbus.Byte(0>
```

Brightness works without problem, but it always show write light. Any ideas? thanks. =D

z3ntu, 2018-05-22:
Is this still a problem?

zeGrandpa, 2018-05-22:
Not for me on arch at least.

z3ntu, 2018-05-22:
@zeGrandpa You also have the same device (so Razer Blade Stealth Mid 2017)?

zeGrandpa, 2018-05-22:
I have the late 2017 one ( that's what the title says )

z3ntu, 2018-05-22:
Okay thanks for telling me :)