openrazer/openrazer #565: Blackwidow Stealth 2014 official support

leandroluiz, 2018-04-04:
Hi everyone,

First, I would like to say thanks for the awesome work on this driver. I'm using a Blackwidow 2014 Stealth and is detected as Blackwidow Classic by the driver and it's working just fine, all keys working, macros on the fly working, macro keys working, that's what I use. But I would like to contribute to make it officially supported and added to the supported devices list. Thanks in advance and let me know if you need more information.

lsusb -d 1532: -v 

> Bus 002 Device 003: ID 1532:011b Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x011b 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
Device Status:     0x0000
  (Bus Powered)

ls -lah /dev/input/by-id/

> total 0
drwxr-xr-x 2 root root 200 Abr  3 21:41 .
drwxr-xr-x 4 root root 500 Abr  3 21:41 ..
lrwxrwxrwx 1 root root  10 Abr  3 21:41 usb-AAB3J17HR_1.3M_HD_WebCam-event-if00 -> ../event13
lrwxrwxrwx 1 root root   9 Abr  3 21:41 usb-Dell_Dell_Universal_Receiver-event-kbd -> ../event9
lrwxrwxrwx 1 root root  10 Abr  3 21:41 usb-Dell_Dell_Universal_Receiver-if01-event-mouse -> ../event10
lrwxrwxrwx 1 root root   9 Abr  3 21:41 usb-Dell_Dell_Universal_Receiver-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Abr  3 21:41 usb-Razer_Razer_BlackWidow-event-kbd -> ../event14
lrwxrwxrwx 1 root root   9 Abr  3 21:41 usb-Razer_Razer_BlackWidow-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Abr  3 21:41 usb-Razer_Razer_BlackWidow-if02-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Abr  3 21:41 usb-Razer_Razer_BlackWidow-if02-mouse -> ../mouse0

Model #: RZ03-0039
Product #: RZ03-00393600-R3M1

z3ntu, 2018-04-04:
Do you have a screenshot from Razer Synapse with your device connected? Is it possible that the "Classic" is the Stealth 2014 or the other way around?
Also see https://github.com/pez2001/razer_chroma_drivers/issues/89#issuecomment-220859877

leandroluiz, 2018-04-04:
Hi again,

I'm pretty sure that Blackwidow Classic is Blackwidow Stealth 2014. I've installed polychromatic and I can toggle gaming mode and change logo brightness and polychromatic identifies it as a blackwidow classic. Do you still need Synapse screenshot? Let me know if you need more tests or anything else. Thanks again.

z3ntu, 2018-04-04:
Polychromatic says it's a BlackWidow Classic as the daemon tells the application so.
A Synapse screenshot would be nice to have confirmation of the name Synapse uses for this exact device.

leandroluiz, 2018-04-06:
Synapse screenshot attached below as your asked. Hope it helps. Thanks again.

[image]

z3ntu, 2018-04-06:
So everything except the device name is correct in the driver? I'll change the name then.

commit 250b27a3 (2018-04-06): Apparently this device is actually called Razer BlackWidow Stealth

Fixes #565
closed 2018-04-06 by z3ntu