openrazer/openrazer #568: Add device support for Razer Kraken 7.1 Classic. 1532:0506

MasterUK, 2018-04-09:
So for some reason my Razer Kraken 7.1 Classic headset does not show up as the same number as in the list. Maybe this is because I am from the UK... Could you please assist me in getting this to work.

z3ntu, 2018-04-09:
* Please post the output of
`lsusb -d 1532:0506 -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

MasterUK, 2018-04-09:
Heading off to bed. Will comment with the info you have requested tomorrow.
Thanks for the fast response.

MasterUK, 2018-04-10:
Here are the results from the first two commands.
Going to switch to windows for the screenshots and read that links content.
[lsusb.txt](https://github.com/openrazer/openrazer/files/1892655/lsusb.txt)
[by-id.txt](https://github.com/openrazer/openrazer/files/1892657/by-id.txt)

MasterUK, 2018-04-10:
[image]
[image]
[image]
[image]
[image]
[image]

z3ntu, 2018-04-10:
Which distro do you use? I have a driver that might work.

MasterUK, 2018-04-10:
Lubuntu.

z3ntu, 2018-04-10:
Please follow https://github.com/openrazer/openrazer/wiki/Building-a-package#ubuntu--linux-mint and replace the `TARGET_BRANCH` with `feature_kraken71_2`

MasterUK, 2018-04-10:
Will try after work tomorrow. Thanks.

MasterUK, 2018-07-18:
Oh S##t, Forgot about this. Will try them this weekend or tomorrow if I get a bit of free time.

commit d892535a (2018-12-30): Apparently Razer Kraken 7.1 0x0506 is like 0x0501

Fixes #568, closes #733
closed 2018-12-30 by z3ntu