openrazer/openrazer #569: REQUEST: Support Razer Imperator Model RZ01-0035

macvelli, 2018-04-10:
I installed the OpenRazer drivers and razerGenie on an Ubuntu 17.10 Desktop machine and my Razer Imperator Model RZ01-0035 is not being recognized. The device support list says the Razer Imperator 2012 is supported, but I am pretty sure mine is the previous model before then.

There is a [Windows driver for my mouse](http://drivers.razersupport.com/index.php?_m=downloads&_a=viewdownload&downloaditemid=475&nav=0) so hopefully it is no big deal to add it to the list of supported devices under Linux.

[image]

z3ntu, 2018-04-10:
`lsusb` output please.

macvelli, 2018-04-10:
foo@bar:~$ lsusb
Bus 006 Device 002: ID 1058:25ee Western Digital Technologies, Inc. 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1532:0017 Razer USA, Ltd RZ01-0035 Laser Gaming Mouse [Imperator]
Bus 001 Device 003: ID 04f2:0833 Chicony Electronics Co., Ltd 
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
foo@bar:~$

z3ntu, 2018-04-10:
Duplicate of #488

closed 2018-04-10 by z3ntu