openrazer/openrazer #571: Razer Naga Hex v2 not recognised

anark10n, 2018-04-15:
Hey there
I'm running an Arch system (4.15) and openrazer installed with polychromatic. Polychromatic is reporting that the openrazer-daemon has detected the razer device but that the device is not recognized, and recommending i bind the driver to device. I looked up where to find the razer driver, but it does not appear in any of the directories when i check them. I followed the [these](https://github.com/openrazer/openrazer/issues/106) instructions in the hopes it would get it working, but I still get the same error message.
Some help with this would be much appreciated, as I am not to familiar with this procedure of binding drivers to devices.

z3ntu, 2018-04-15:
Please post the output of `sudo dkms status openrazer-driver/1.0.0` and `openrazer-daemon -Fv`

anark10n, 2018-04-15:
`sudo dkms status openrazer-driver/1.0.0` returns no output.
`openrazer-daemon -Fv` returns
```Starting daemon.
2018-04-15 11:56:33 | razer                          | INFO     | Initialising Daemon (v2.3.0). Pid: 2106
2018-04-15 11:56:33 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2018-04-15 11:56:33 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
[... cut ...]
2018-04-15 11:56:33 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
2018-04-15 11:56:33 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
2018-04-15 11:56:33 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2018-04-15 11:56:33 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
2018-04-15 11:56:33 | razer                          | INFO     | Serving DBus```

Just hangs on that last line.

z3ntu, 2018-04-15:
Then run `sudo dkms install openrazer-driver/1.0.0`

anark10n, 2018-04-15:
`sudo dkms install openrazer-driver/1.0.0` produces the following output

```
Error! Could not find module source directory.
Directory: /usr/src/openrazer-driver-1.0.0 does not exist.
```

anark10n, 2018-04-15:
`pacman -Q | grep razer` returns
```
openrazer-daemon-git 2.3.0.r8.gaf10903-1
openrazer-driver-dkms-git 2.3.0.r8.gaf10903-1
openrazer-meta-git 2.3.0.r8.gaf10903-1
python-openrazer-git 2.3.0.r8.gaf10903-1
```

z3ntu, 2018-04-15:
Ah you're using the -git package. Then it's `sudo dkms install openrazer-driver/2.3.0`

anark10n, 2018-04-15:
Ok got it installed, output of `sudo dkms install openrazer-driver/2.3.0` returns
```
Module openrazer-driver/2.3.0 already installed on kernel 4.15.15-1-ARCH/x86_64
```

anark10n, 2018-04-15:
Ok, a restart sorted out all the issues. Appreciate the assist.