openrazer/openrazer #572: Razer Anansi macro issue

lccy, 2018-04-15:
Hi,

I am currently running arch linux and using razer anansi keyboard.
Have installed openrazer-meta and tried to setup macro.
Below are few observations:
1) FN+macro (right alt key) doesnt work
2) macro M4 is enabling macro recording
3) macro M5 is toggling gaming mode
4) only macro M1, M2 and M3 are assignable, the thumb macro buttons are not recognized

z3ntu, 2018-05-22:
This seems to be related to #577

closed 2018-10-19 by stale[bot]

lehesepp, 2019-01-10:
I too have the exact issue on xubuntu 18.04. What's the fix?

z3ntu, 2019-01-15:
@lehesepp The above linked issue was fixed with version 2.3.1, so using that or a newer version should be good

lehesepp, 2019-01-16:
Thanks for replying. However i just recently installed the openrazer toolset and my Razer Anansi is still experiencing issues mentioned in this topic's original post :( I have openrazer daemon and Razer python lib both with version 2.4.0 using Polychromatic controller v. 0.3.12 stable. Just recently bought this keyboard however my computer is very old. Also confirmed than in windows for different computer the macro keys work fine.

Also I don't believe the linked issue is the issue described here nor is it for the Razer Anansi keyboard.

NothingClever, 2020-04-29:
I'm using Fedora 31, Daemon version 2.7.0, with an Anansi, firmware 1.0, and I just had the same issue. I did some digging and found that I was getting the following key_ids:
M1: 185, M2: 186, M3: 187, M4: 188, M5: 189
T1: 190, T2: 191, T3: 192, T4: 193, T5: 90, T6: 91, T7: 95

The Fn+F9 combination for recording macros and the Fn+F12 combination for brightness are also broken but I didn't bother looking into it since I remapped those functions to the thumb keys. Fn+F11 works for setting gamemode, which is correct for the label on the keyboard.

If someone is desperate, I fixed the issue by changing the keys in the EVENT_MAPPING dictionary to match the correct key codes. The thumb keys don't have any special functions so I mapped them to the macromode, gamemode, and brightness events. I edited line 138 of keyboard.py. For me it was located at /usr/lib/python3.7/site-packages/openrazer_daemon/ Of course, this will mess up other keyboards, won't persist if there is an update without a fix, and requires restarting the daemon.