openrazer/openrazer #574: Can't start openrazer-daemon on Mint 18.3

DanyGee, 2018-04-18:
**razercommander**
Importing device logic, waiting for daemon
ERROR: the daemon is not responding!
Try running `killall razer-daemon && razer-daemon` or rebooting. If this doesn't work, please fill an issue!
Exception: org.freedesktop.DBus.Python.FileNotFoundError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/all.py", line 85, in get_keyboard_layout
    with open(driver_path, 'r') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:1c.4/0000:0d:00.0/usb5/5-2/5-2:1.2/0003:1532:021A.0006/kbd_layout'

**lsmod | grep razer**
razerkbd               57344  0
hid                   118784  7 i2c_hid,hid_generic,usbhid,hid_logitech_dj,razerkbd,hid_logitech_hidpp

**sudo journalctl -b | grep -i razer_mount**        
kwi 18 15:49:38 linuxmint razer_mount[810]: Driver razerkbd
kwi 18 15:49:38 linuxmint razer_mount[809]: Driver razerkbd
kwi 18 15:49:38 linuxmint razer_mount[812]: Device_ID 0003:1532:021A.0005
kwi 18 15:49:38 linuxmint razer_mount[814]: Unbinding 0003:1532:021A.0005 from hid-generic
kwi 18 15:49:38 linuxmint razer_mount[813]: Device_ID 0003:1532:021A.0004
kwi 18 15:49:38 linuxmint razer_mount[815]: Unbinding 0003:1532:021A.0004 from hid-generic
kwi 18 15:49:38 linuxmint razer_mount[838]: Driver razerkbd
kwi 18 15:49:38 linuxmint razer_mount[841]: Device_ID 0003:1532:021A.0006
kwi 18 15:49:38 linuxmint razer_mount[846]: Unbinding 0003:1532:021A.0006 from hid-generic
kwi 18 15:49:38 linuxmint razer_mount[1353]: Binding 0003:1532:021A.0005 to razerkbd
kwi 18 15:49:39 linuxmint razer_mount[1372]: Binding 0003:1532:021A.0004 to razerkbd
kwi 18 15:49:39 linuxmint razer_mount[1373]: Binding 0003:1532:021A.0006 to razerkbd
kwi 18 15:49:39 linuxmint razer_mount[1392]: Finished binding 0003:1532:021A.0005
kwi 18 15:49:39 linuxmint razer_mount[1393]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:021A.0005/
kwi 18 15:49:39 linuxmint razer_mount[1395]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:021A.0005/
kwi 18 15:49:39 linuxmint razer_mount[1413]: Finished binding 0003:1532:021A.0004
kwi 18 15:49:39 linuxmint razer_mount[1414]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:021A.0004/
kwi 18 15:49:39 linuxmint razer_mount[1416]: Finished binding 0003:1532:021A.0006
kwi 18 15:49:39 linuxmint razer_mount[1417]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:021A.0006/
kwi 18 15:49:39 linuxmint razer_mount[1418]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:021A.0004/
kwi 18 15:49:39 linuxmint razer_mount[1420]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:021A.0006/

**System**
OS: Linux Mint 18.3 Sylvia x86_64 
Kernel: 4.13.0-36-generic 
Uptime: 48 mins 
Packages: 3229 
Shell: bash 4.3.48 
Resolution: 3840x2160 
DE: Cinnamon 3.6.7 
WM: Mutter (Muffin) 
WM Theme: Arc-Dark (Arc-Dark) 
Theme: Arc-Dark [GTK2/3] 
Icons: Papirus-Dark [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel i7-6700K (8) @ 4.300GHz 
GPU: NVIDIA NVIDIA Corporation Device 1b80
Memory: 2617MiB / 32102MiB 

If you need more info - ask.

z3ntu, 2018-04-18:
Please run `find /usr/lib/modules -name "razer*.ko"`, and send me the output.

If you just want to fix it you can then follow https://github.com/polychromatic/polychromatic/issues/146#issuecomment-379465409 otherwise we could try to figure out why this weird behaviour happens (i.e. kernel module not getting updated or something).

DanyGee, 2018-04-18:
**find /usr/lib/modules -name "razer*.ko"**
find: `/usr/lib/modules': no such file or directory

z3ntu, 2018-04-18:
`sudo dkms remove openrazer-driver/1.0.0 --all` and then the find command again?

DanyGee, 2018-04-18:
> sudo dkms remove openrazer-driver/1.0.0 --all and then the find command again?

Now the find command finds nothing.

z3ntu, 2018-04-18:
`sudo dkms install openrazer-driver/1.0.0` and reboot. Does it work then?

DanyGee, 2018-04-18:
> sudo dkms install openrazer-driver/1.0.0 and reboot. Does it work then?

**razercommander**
Importing device logic, waiting for daemon
ERROR: the daemon is not responding!
Try running `killall razer-daemon && razer-daemon` or rebooting. If this doesn't work, please fill an issue!
Exception: org.freedesktop.DBus.Python.FileNotFoundError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/all.py", line 85, in get_keyboard_layout
    with open(driver_path, 'r') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:1c.4/0000:0d:00.0/usb5/5-2/5-2:1.2/0003:1532:021A.0006/kbd_layout'

z3ntu, 2018-04-18:
```
dpkg -l
sudo dkms remove openrazer-driver/1.0.0 --all
sudo mv /usr/src/openrazer-driver-1.0.0 /usr/src/openrazer-driver_old-1.0.0
sudo apt install --reinstall openrazer-kernel-modules-dkms
```
And then reboot (and the output of the dpkg command before please)

DanyGee, 2018-04-18:
Same deal.

**razercommander**
Importing device logic, waiting for daemon
ERROR: the daemon is not responding!
Try running `killall razer-daemon && razer-daemon` or rebooting. If this doesn't work, please fill an issue!
Exception: org.freedesktop.DBus.Python.FileNotFoundError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/dbus_services/dbus_methods/all.py", line 85, in get_keyboard_layout
    with open(driver_path, 'r') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:1c.4/0000:0d:00.0/usb5/5-2/5-2:1.2/0003:1532:021A.0006/kbd_layout'

[dpkg_and_others.log](https://github.com/openrazer/openrazer/files/1924818/dpkg_and_others.log)

closed 2018-04-18 by z3ntu