openrazer/openrazer #576: Razer Mamba (2012) Wireless Constant battery warnings

wolfen351, 2018-04-21:
I have been having regular incorrect battery warnings for the mouse forever. I am hoping that there is a simple switch to turn the battery warnings off, or something that I can do to help the issue be corrected.

This is with the mouse battery practically full:
`
...
cat ~/.local/share/openrazer/logs/razer.log | grep -a battery
2018-04-21 12:59:41 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 13:09:42 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 13:09:42 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 13:19:43 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 13:19:43 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 13:29:44 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 13:29:44 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 13:39:45 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 13:39:45 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 13:49:46 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 13:49:46 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 13:59:47 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 13:59:47 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 14:09:48 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 14:09:48 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 14:19:49 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 14:19:49 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 14:29:50 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 14:29:50 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 14:39:51 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 14:39:51 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 14:49:52 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 14:49:53 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 100.0%
2018-04-21 14:59:54 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 14:59:54 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 2.7%
...
`

wolfen351, 2018-04-21:
Here is a snippet from the end of dmesg:

`[60720.603789] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[61321.629458] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[61922.647118] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[62523.687893] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[63124.688814] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[63725.677878] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[64326.666526] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[64927.739321] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[65528.727235] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[66129.780421] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[66730.767588] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[67331.758192] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[67932.835040] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[68533.824795] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[69134.816964] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[69735.808737] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[70336.795259] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[70937.797102] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[71538.779082] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[72139.763579] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[72740.759667] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[73341.740479] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[73942.728811] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[74543.719610] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[75144.715380] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[75745.774489] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[76346.762557] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[76947.751970] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[77548.735872] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[78149.716446] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[78750.703964] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[79351.745924] razermouse: Command timed out. Start Marker: 04 id: ff Num Params: 02 Reserved: 07 Command: 80 Params: 00000000000000000000000000000000 .
[80680.171468] razermouse: Response doesnt match request. Start Marker: 01 id: ff Num Params: 02 Reserved: 00 Command: 81 Params: 00000000000000000000000000000000 .
[81154.837936] razermouse: Response doesnt match request. Start Marker: 02 id: ff Num Params: 02 Reserved: 00 Command: 81 Params: 01070000000000000000000000000000 .
`

wolfen351, 2018-04-21:
here are the last 100 log lines:

`wolfen@wolfen-HP:~$ tail  -n 100 ~/.local/share/openrazer/logs/razer.log
2018-04-21 09:19:19 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 09:19:19 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 09:29:20 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 09:29:20 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 09:39:21 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 09:39:21 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 09:49:22 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 09:49:22 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 09:59:23 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 09:59:23 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 10:04:20 | razer.screensaver              | DEBUG    | Received screensaver active signal
2018-04-21 10:04:20 | razer.device0                  | INFO     | Suspending RazerBlackWidowUltimate2013
2018-04-21 10:04:20 | razer.device0                  | DEBUG    | DBus call get_brightness
2018-04-21 10:04:20 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-04-21 10:04:20 | razer.device1                  | INFO     | Suspending RazerMamba2012Wireless
2018-04-21 10:04:20 | razer.device1                  | DEBUG    | DBus call get_brightness
2018-04-21 10:04:51 | razer.screensaver              | DEBUG    | Received screensaver inactive signal
2018-04-21 10:04:51 | razer.device0                  | INFO     | Resuming RazerBlackWidowUltimate2013
2018-04-21 10:04:51 | razer.device0                  | DEBUG    | DBus call set_brightness
2018-04-21 10:04:51 | razer.device1                  | INFO     | Resuming RazerMamba2012Wireless
2018-04-21 10:04:51 | razer.device1                  | DEBUG    | DBus call set_brightness
2018-04-21 10:09:24 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 10:09:24 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 10:19:25 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 10:19:25 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 10:29:26 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 10:29:26 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 10:39:27 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 10:39:27 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 10:49:28 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 10:49:28 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 10:59:29 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 10:59:29 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 11:09:30 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 11:09:30 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 11:19:31 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 11:19:31 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 11:29:32 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 11:29:32 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 11:39:33 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 11:39:33 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 11:49:34 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 11:49:34 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 11:59:35 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 11:59:35 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 12:09:36 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 12:09:36 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 12:19:37 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 12:19:37 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 12:29:38 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 12:29:38 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 12:39:39 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 12:39:39 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 12:49:40 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 12:49:40 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 12:59:41 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 12:59:41 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
2018-04-21 13:09:42 | razer.device1                  | DEBUG    | DBus call get_battery
[... cut ...]
2018-04-21 14:51:58 | razer.device1                  | DEBUG    | DBus call has_matrix
2018-04-21 14:51:59 | razer.device1                  | DEBUG    | DBus call get_scroll_active
2018-04-21 14:59:54 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 14:59:54 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 2.7%
2018-04-21 15:09:55 | razer.device1                  | DEBUG    | DBus call get_battery
2018-04-21 15:09:55 | razer.device1.batterynotifier  | DEBUG    | Razer Mamba Battery at 0.0%
`

wolfen351, 2018-04-21:
Here is razer mount:

`wolfen@wolfen-HP:~$ sudo journalctl -b | grep -i razer_mount
Apr 20 16:27:22 wolfen-HP razer_mount[1736]: Driver razermouse
Apr 20 16:27:22 wolfen-HP razer_mount[1737]: Driver razerkbd
Apr 20 16:27:22 wolfen-HP razer_mount[1734]: Driver razerkbd
Apr 20 16:27:22 wolfen-HP razer_mount[1735]: Driver razermouse
Apr 20 16:27:22 wolfen-HP razer_mount[1738]: Driver razerkbd
Apr 20 16:27:22 wolfen-HP razer_mount[1739]: Device_ID 0003:1532:0025.0006
Apr 20 16:27:22 wolfen-HP razer_mount[1740]: Device_ID 0003:1532:011A.0004
Apr 20 16:27:22 wolfen-HP razer_mount[1741]: Device_ID 0003:1532:0025.0005
Apr 20 16:27:22 wolfen-HP razer_mount[1744]: Unbinding 0003:1532:0025.0006 from hid-generic
Apr 20 16:27:22 wolfen-HP razer_mount[1742]: Device_ID 0003:1532:011A.0002
Apr 20 16:27:22 wolfen-HP razer_mount[1745]: Unbinding 0003:1532:011A.0004 from hid-generic
Apr 20 16:27:22 wolfen-HP razer_mount[1746]: Unbinding 0003:1532:0025.0005 from hid-generic
Apr 20 16:27:22 wolfen-HP razer_mount[1743]: Device_ID 0003:1532:011A.0003
Apr 20 16:27:22 wolfen-HP razer_mount[1747]: Unbinding 0003:1532:011A.0002 from hid-generic
Apr 20 16:27:22 wolfen-HP razer_mount[1748]: Unbinding 0003:1532:011A.0003 from hid-generic
Apr 20 16:27:22 wolfen-HP razer_mount[1758]: Binding 0003:1532:0025.0006 to razermouse
Apr 20 16:27:22 wolfen-HP razer_mount[1761]: Binding 0003:1532:0025.0005 to razermouse
Apr 20 16:27:22 wolfen-HP razer_mount[1782]: Binding 0003:1532:011A.0002 to razerkbd
Apr 20 16:27:22 wolfen-HP razer_mount[1789]: Binding 0003:1532:011A.0003 to razerkbd
Apr 20 16:27:22 wolfen-HP razer_mount[1790]: Binding 0003:1532:011A.0004 to razerkbd
Apr 20 16:27:22 wolfen-HP razer_mount[1807]: Finished binding 0003:1532:0025.0006
Apr 20 16:27:22 wolfen-HP razer_mount[1808]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0025.0006/
Apr 20 16:27:22 wolfen-HP razer_mount[1810]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0025.0006/
Apr 20 16:27:23 wolfen-HP razer_mount[1813]: Finished binding 0003:1532:0025.0005
Apr 20 16:27:23 wolfen-HP razer_mount[1814]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0025.0005/
Apr 20 16:27:23 wolfen-HP razer_mount[1816]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0025.0005/
Apr 20 16:27:23 wolfen-HP razer_mount[1827]: Finished binding 0003:1532:011A.0002
Apr 20 16:27:23 wolfen-HP razer_mount[1828]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0002/
Apr 20 16:27:23 wolfen-HP razer_mount[1830]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0002/
Apr 20 16:27:23 wolfen-HP razer_mount[1831]: Finished binding 0003:1532:011A.0003
Apr 20 16:27:23 wolfen-HP razer_mount[1832]: Finished binding 0003:1532:011A.0004
Apr 20 16:27:23 wolfen-HP razer_mount[1835]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0003/
Apr 20 16:27:23 wolfen-HP razer_mount[1834]: Changing group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0004/
Apr 20 16:27:23 wolfen-HP razer_mount[1838]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0003/
Apr 20 16:27:23 wolfen-HP razer_mount[1839]: Changed group /sys/bus/hid/drivers/razerkbd/0003:1532:011A.0004/
`

wolfen351, 2018-04-21:
Here are some related tickets from the polychromatic UI:

https://github.com/polychromatic/polychromatic/issues/106
https://github.com/polychromatic/polychromatic/issues/147

wolfen351, 2018-04-21:
Ok thats all the info I have - please let me know if there is something more I can do to help debug/fix this issue :)

wolfen351, 2018-04-21:
Ok, I think I have one more piece if info.. perhaps all the popup notifications are cached while the pc is locked, because I just unlocked it now and I get essentially continuous notifications with different % in them.. mainly 0% and 100% at the moment..   This is what originally prompted me to log this ticket.

wolfen351, 2018-09-18:
But but.. *sigh*

lah7, 2018-09-19:
No stalebot, keep this one open from a UI perspective! :robot:

I suppose what's needed is some extra config options in `~/.config/openrazer/razer.conf`, **or** even better, expose it in the Python library to offload the notification responsibility to front-ends.

nshiell, 2020-05-07:
It would be good if there was an option to only show battery warnings when they fall below a value.
In the same way as my phone notifies me when it's about to run out of charge.

mgbowe1, 2022-02-04:
Would love if there was a configurable battery percentage where a notification was sent. I'm tired of getting notifications every time the percentage is queried.

lah7, 2022-02-04:
I made improvements to the battery notification which is now in the `master` branch (#1666). The default now is to only show when it reaches 33%, but you can configure this with `battery_notifier_percent` in `~/.config/openrazer/razer.conf`.

If you'd like this early, consider installing the daily builds (on Ubuntu) or `-git` package (on Arch)

closed 2022-02-04 by lah7