openrazer/openrazer #577: Chroma keys transalation screws Blackwidow Multimedia Keys

myelsukov, 2018-04-21:
Could you please check the keyboard model prior to translation that uses this table? https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L56

On my Blackwidow keyboard multimedia keys layout is different:
F5 - PlayPause
F6 - Stop
F7 - Prev
F8 - Next

z3ntu, 2018-04-27:
`lsusb | grep 1532` please

myelsukov, 2018-05-06:
```
>lsusb | grep 1532
Bus 001 Device 006: ID 1532:010e Razer USA, Ltd
```
Keyboard Layout:
F1 - Mute
F2 - Volume Down
F3 - Volume Up
F5 - Play/Pause
F6 - Stop Playback
F7 - Previous Track
F8 - Next Track
F12 - Brightness
Pause/Preak - Sleep
Right Alt - Razer Macro

To make it work properly on my keyboard I had to modify this table: https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L56

```
static const struct razer_key_translation chroma_keys[] = {
    { KEY_F1,    KEY_MUTE },
    { KEY_F2,    KEY_VOLUMEDOWN },
    { KEY_F3,    KEY_VOLUMEUP },

    { KEY_F7,    KEY_PREVIOUSSONG },
    { KEY_F5,    KEY_PLAYPAUSE },
    { KEY_F8,    KEY_NEXTSONG },
    { KEY_RIGHTALT,    RAZER_MACRO_KEY },

    { KEY_PAUSE, KEY_SLEEP },
    { }
};

```

terrycain, 2018-05-06:
@myelsukov Remind me Saturday and I'll have a look, unless @z3ntu fixed it

z3ntu, 2018-05-20:
Can you please try the commit 0b806965a7c17ff6461eabb8db4d53313178cce2

myelsukov, 2018-05-20:
This works.
I am sorry, but I forgot to mention two other keys:

F11 - Game Mode
F12 - Brightness. I have just a single brightness button. Never used it :)

z3ntu, 2018-05-20:
First, does your keyboard have the same (special) keys as https://d1urewwzb2qwii.cloudfront.net/sys-master/images/h7c/h12/8823290396702 ? Second, is this keyboard really called "BlackWidow Classic"? Does the sticker on the bottom say that?
And third, do you know what that brightness button does on Windows with Synapse?

myelsukov, 2018-05-20:
It looks exactly like my keyboard.
The sticker says: "Blackwidow Stealth Edition"

z3ntu, 2018-05-20:
Okay I found 

>  as well control backlighting (F12). The backlight has 5 modes and varies the intensity of the blue LED with last function being a pulsating backlight.

on https://www.bjorn3d.com/2011/10/razer-blackwidow-ultimate-stealth-edition/ which seems to be a review of the Ultimate equivalent of your keyboard. Is yours also from about 2011? Apparently everything pre-2013 was banished from the Razer website never to be found ^^

And I found the product page probably for your keyboard on archive.org: https://web.archive.org/web/20110924152039/http://store.razerzone.com:80/store/razerusa/en_US/pd/productID.235228400/categoryId.49136200/parentCategoryId.35156900 
I'll change the name but I think I am leaving that 'backlighting control key' unmapped for now as nothing is implemented on the daemon side to handle that.

myelsukov, 2018-05-20:
I don't remember exactly when I got this keyboard but it was long enough ago :)
And I couldn't care less about backlight feature (gaming mode too :) ).

z3ntu, 2018-05-20:
Try again with the latest commit on the branch `010e_translation` :)

myelsukov, 2018-05-20:
Cool! Thanks! I am not home now. Will test it later.

myelsukov, 2018-05-22:
This works better. I missed that break too :)

myelsukov, 2018-05-30:
Yes. With "break" it works as expected.

commit 0b806965 (2018-05-20): Fix chroma key translation on 010e

Fixes #577
commit cec08420 (2018-05-30): Fix chroma key translation on 0x010e

Also rename the device to "BlackWidow Stealth Edition"

Fixes #577
closed 2018-05-30 by z3ntu