openrazer/openrazer #578: Razergenie on arch linux

pranayama26, 2018-04-23:
`● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-04-22 09:55:03 CDT; 24s ago
     Docs: man:openrazer-daemon(8)
  Process: 21914 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 21914 (code=exited, status=1/FAILURE)

Apr 22 09:55:03 anarchy systemd[507]: Starting Daemon to manage razer devices in userspace...
Apr 22 09:55:03 anarchy openrazer-daemon[21914]: Starting daemon.
Apr 22 09:55:03 anarchy openrazer-daemon[21914]: 2018-04-22 09:55:03 | razer                          | CRITICAL | User is not a member of the plugdev group
Apr 22 09:55:03 anarchy openrazer-daemon[21914]: Stopping daemon.
Apr 22 09:55:03 anarchy systemd[507]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Apr 22 09:55:03 anarchy systemd[507]: openrazer-daemon.service: Failed with result 'exit-code'.
Apr 22 09:55:03 anarchy systemd[507]: Failed to start Daemon to manage razer devices in userspace.

`

lah7, 2018-04-23:
You'll need to add your user to the plugdev group, then log out and back in:

    sudo gpasswd -a <YOUR-USERNAME> plugdev

closed 2018-05-05 by z3ntu

prochac, 2021-03-24:
Would be nice to know Why? it's needed :)

z3ntu, 2021-03-24:
Because the daemon runs as your user and we do permissions on the sysfs files via this plugdev group, so your user needs to be in the group to be able to read and write to the sysfs files.