openrazer/openrazer #583: Razer BlackWidow Chroma V2 changes color of some number keys (perhaps on suspend)

nightsky30, 2018-05-01:
My Razer BlackWidow Chroma V2 was functioning very well, and still does for the most part. I recently updated to Ubuntu 18.04, and have noticed 2 or 3 times now over the past few days that some of my number pad keys change to a green color.

- I have reproduced this multiple times.
- I believe it is related to my system going into suspend mode.
- My profile is simply a static orange hue.
- The affected keys are always the default green no matter the static color selected for the rest of the keyboard.
- The keys all start out the correct color at boot and remain lit in the proper color while in use.
- They change to green, and it is always the same keys that change:

- 0, 2, 5, 6, 8, 9, /, AND *

I ran "sudo openrazer-daemon -Fv --as-root", and let my system suspend.  When I came back the above numpad keys were green, and there was nothing strange in the console logs.  In fact, they hadn't changed a bit since before the suspend ~3hrs of use / wait to suspend.

dmesg doesn't show anything strange that I can see.  There are some USB disconnects as I was trying to reproduce the color change by removing and reinserting my mouse at first.  I then played a few games and finally let my system suspend.  After the suspend is when it changed color.

[daemon.txt](https://github.com/openrazer/openrazer/files/1962844/daemon.txt)

[dmesg.txt](https://github.com/openrazer/openrazer/files/1962857/dmesg.txt)

~$ uname -a
Linux 4.15.0-20-generic #21-Ubuntu SMP Tue Apr 24 06:16:15 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04 LTS
Release:	18.04
Codename:	bionic

nightsky30, 2018-05-01:
If there are other logs I can grab, let me know.

z3ntu, 2018-05-01:
Does this also happen without the daemon running?

nightsky30, 2018-05-04:
It seems to happen after being in suspend a while, but not immediately when suspend happens.  It didn't happen the other day when I turned the daemon off, but maybe I hadn't waited long enough after the suspend for the problem to trigger.

I also think it might be a firmware issue.  This is what it looks like, and this person runs Windows:
https://youtu.be/pdSpT-_77jo?t=32

They linked to a firmware executable in the comments section.  I'm apprehensive about downloading and running strange executables which aren't readily available on razer's site.  I might contact support to see what they have.  I don't dual boot, but I have ways around the Windows specific firmware if this will indeed fix the issue.

nightsky30, 2018-05-04:
Another example:  https://www.youtube.com/watch?v=fL9c_qx9Zc0

I contacted support.  I'm going to close this as it does not appear to be an issue with the daemon.  Thank you.

closed 2018-05-04 by nightsky30

terrycain, 2018-05-04:
@nightsky30 👍

nightsky30, 2018-05-31:
FYI, just wanted to follow up and note that the firmware update I received from support fixed the strange lighting issues for me.  It is a pity they do not supply firmware downloads on the razer site.