openrazer/openrazer #588: Build for Fedora 28

AbsyntheSyne, 2018-05-07:
Hi. I recently installed Fedora 28 on my desktop and there's no repository for 28, but there is for 27, 26, and rawhide. Is there anyway you could get binary rpm packages built for 28 as well? Thanks!

z3ntu, 2018-05-07:
I've just enabled the repository and packages should be built and published over the next couple of hours (hopefully).

z3ntu, 2018-05-07:
You can view the status at https://build.opensuse.org/project/monitor/hardware:razer?arch_x86_64=1&blocked=1&broken=1&building=1&defaults=0&deleting=1&dispatching=1&failed=1&finished=1&locked=1&repo_Fedora_28=1&scheduled=1&signing=1&succeeded=1&unresolvable=1

closed 2018-05-07 by z3ntu

AbsyntheSyne, 2018-05-07:
Oh awesome! Thanks! :D

z3ntu, 2018-05-07:
Repo should be published. Have fun!

kierun, 2018-06-11:
Could someone update the documentation? Currently, it appears only 26, 27, and rawhide are supported. [Reference](https://openrazer.github.io/#fedora)

z3ntu, 2018-06-11:
Fixed in https://github.com/openrazer/openrazer.github.io/commit/e15e74bfad35bc12e9dc3e4f28e33533c75c23dd