openrazer/openrazer #590: Changing Macros on the Naga Hex v2

MonkeyDStyxx, 2018-05-08:
So i just bought a new mouse (finally after 7+ years of my old one) and i have installed all the drivers and programs i need, but i am trying to change my key bindings and macros for the buttons on the mouse and i am unsure how to do that. not sure if i missed something in the programs i have (razer genie/polychromatic controller) or if i am just unable to change them. 

any help is appreciated! TIA

Altoholic, 2018-10-02:
Any news on that ?
Is the Naga Hex v2 still supported ?

fang0654, 2019-02-05:
Thank you folks for posting this. After uncommenting the lines in both mouse.py (for openrazer) and devicesMacroKeys.py (for razercommander) I have full macro support on my Naga Hex V2! I don't know why it was disabled, but if you re-enable it, it works well.

closed 2019-03-07 by stale[bot]