openrazer/openrazer #591: Keys get stuck randomly

Berger92, 2018-05-11:
Hi Guys,

I have a Razer Ornata Chroma. I had an issue previously, then it went away for 3-4 months. Now it is back.  I am running Ubuntu 16.04 server. Some keys get stuck randomly. Most cases it is the Ctrl. The key itself doesn't get stuck but it keeps writing letters like this: ccccccccccccccccccccccccccccccccccccccc

I have no clue the what the problem can be, but it definately worked for months flawlessly. I have a feeling it is the driver/deamon and not the keyboard itself.

Any ideas would be appreciated.

z3ntu, 2018-05-11:
Does it still happen with the driver&daemom uninstalled?

terrycain, 2018-05-11:
I rarely see this. Happens at most once every two months for me, is
possibly the driver, I've had a look through the code and can't find
anything that would cause it so if it is the driver, its probably not going
to get fixed by me :D

nightsky30, 2018-05-31:
@Berger92 I had a lighting issue with my keyboard (BlackWidow Chroma v2) which happened very randomly.  I contacted Razer support and they sent me a link to download a firmware update(.exe) which fixed the problem.

I'm not saying that is definitely the case here.  Just wondering if you'd tried contacting support to see if there is a firmware update that might address your issue.  They do not list firmware updates on their site for some reason.  Not very helpful.

Also note, I am running Ubuntu and do not dual boot.  If there is a firmware update, the easiest route might be to borrow a friend's Win PC.  I had to be creative in order to install the firmware update.  It can be done with installation media, a portable USB drive, and [Rufus](https://rufus.akeo.ie/) if you're feeling _**creative**_.

Berger92, 2018-06-04:
I couldn't fix it so far.
@nightsky30 sounds great it is just a bit difficult to do it as I use my keyboard at work.

For now, I just restart the daemon if this happens. It seems to solve the issue most cases.

terrycain, 2018-06-04:
Odd that restarting the daemon fixes it, that shouldn't do any key dodgyness unless you do on-the-fly macro stuff. If you do get a fw upgrade get you firmware version from before and after, least that way we can track the spread of this bug

closed 2018-11-01 by stale[bot]