openrazer/openrazer #592: Nixos support

TheKrister2, 2018-05-13:
Is it possible to get Nixos support?

z3ntu, 2018-05-14:
I am not familiar with NixOS packaging and it doesn't look trivial to do. If you are not willing to do the packaging yourself, maybe you can ask existing NixOS contributors to help out.

TheKrister2, 2018-05-14:
Can't say I'm that familiar with the packaging myself, if I can find any NixOS contributors I'll ask though. Otherwise I guess I'll just have to learn how to do it, though that might take some time.

z3ntu, 2018-10-28:
There's an open PR at https://github.com/NixOS/nixpkgs/pull/47009 . I'm gonna close this issue then.

closed 2018-10-28 by z3ntu

evanjs, 2019-08-29:
Radio silence on NixOS/nixpkgs#47009 so I picked it up in NixOS/nixpkgs#64552. 
I plan on bumping it if/when it‘s merged and then adding it to the install instructions on https://openrazer.github.io