openrazer/openrazer #594: Daemon Not Responding *Daemon Won't Launch at All* *Not working with Polychromatic*

Cutday, 2018-05-18:
I have no idea what I am doing wrong. I was trying to install razer drivers for my mouse and headset

Here's my mouse: Razer Deathadder Chroma (Not sure what year cause I bought it off Ebay) - It's a legit Razer mouse
Here's my headset: Razer Krakens 7.1 Chroma

Now, here's the entire log when I type openrazer-daemon -Fv

2018-05-18 16:56:55 | razer | INFO | Initialising Daemon (v2.3.0). Pid: 1037
2018-05-18 16:56:55 | razer.screensaver | INFO | Initialising DBus Screensaver Monitor
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerKrakenClassic ---------------------- (1532:0501)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerMambaTE ---------------------------- (1532:0046)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerNagaChroma ------------------------- (1532:0053)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerOrbweaverChroma -------------------- (1532:0207)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerOrochi2011 ------------------------- (1532:0013)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerOrochi2013 ------------------------- (1532:0039)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerTaipan ----------------------------- (1532:0034)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerTartarus --------------------------- (1532:0201)
2018-05-18 16:56:55 | razer | DEBUG | Loaded device specification: RazerTartarusChroma --------------------- (1532:0208)
2018-05-18 16:56:55 | razer | INFO | Found device.0: 0003:1532:0504.0003
2018-05-18 16:56:55 | razer.device0 | INFO | Initialising device.0 RazerKraken
[... 32 lines cut ...]
razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/mouse.py", line 249, in init
super(RazerDeathAdderChroma, self).init(*args, **kwargs)
File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 441, in init
super().init(device_path, device_number, config, testing, additional_interfaces, additional_methods)
File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 64, in init
self.serial = self.get_serial()
[... cut ...]
File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 427, in del
self.close()
File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 340, in close
if not self._is_closed:
AttributeError: 'RazerDeathAdderChroma' object has no attribute '_is_closed'
Stopping daemon.

If you know the problem, state the fix for it very specifically! I am awful at using Ubuntu, and I'm still learning so I need to know exactly what I am doing! Thank you!!!!

My OS: Ubuntu 18.04

Cutday, 2018-05-18:
And yes, I have the newest Polychromatic software along with the newest Openrazer drivers, but for some odd reason, the Daemon will just not launch at all.

z3ntu, 2018-05-19:
Can you give us the output of the command `cat /sys/bus/hid/drivers/razermouse/0*/device_serial` ? That endpoint seems to return something the daemon doesn't like.

Cutday, 2018-05-19:
cat /sys/bus/hid/drivers/razermouse/0*/device_serial
����������������������

Cutday, 2018-05-19:
I mean, that's what I got when I ran the command...

z3ntu, 2018-05-19:
What about `cat /sys/bus/hid/drivers/razermouse/0*/device_serial | xxd` ? xxd is provided by vim, so install it if you don't have it.

Cutday, 2018-05-19:
cat /sys/bus/hid/drivers/razermouse/0*/device_serial | xxd
00000000: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00000010: ffff ffff ffff 0a                        .......

Cutday, 2018-05-19:
Can you send me the link to install it please

z3ntu, 2018-05-19:
Please try applying the following patch to the file `/usr/lib/python3.6/site-packages/openrazer_daemon/hardware/device_base.py` or `/lib/python3.*/dist-packages/openrazer_daemon/hardware/device_base.py` dependending on your distro (second path for debian derivatives and the first for the rest)

```diff
diff --git a/daemon/openrazer_daemon/hardware/device_base.py b/daemon/openrazer_daemon/hardware/device_base.py
index 28ebe00..9a35252 100644
--- a/daemon/openrazer_daemon/hardware/device_base.py
+++ b/daemon/openrazer_daemon/hardware/device_base.py
@@ -209,6 +209,9 @@ class RazerDevice(DBusService):
                 except (PermissionError, OSError) as err:
                     self.logger.warning('getting serial: {0}'.format(err))
                     serial = ''
+                except UnicodeDecodeError:
+                    self.logger.warning('malformed serial.')
+                    serial = ''
 
                 count += 1
                 time.sleep(0.1)
```

commit cf4c2bf4 (2018-05-20): Handle UnicodeDecodeError in get_serial

which can get thrown due to malformed serials

Fixes #594 and fixes #558
commit af84bd6c (2018-05-20): Handle UnicodeDecodeError in get_serial

which can get thrown due to malformed serials

Fixes #594 and fixes #558
closed 2018-05-20 by z3ntu