openrazer/openrazer #595: Razer Atheris support

hxss, 2018-05-20:
I want to buy this mouse. Is it possible to configure its buttons with openrazer?
Will my macroses work with bluetooth connection(as I know this mouse don't have internal memory)?

z3ntu, 2018-05-20:
No, this mouse is not yet supported. It seems like a relatively new mouse so support may take a while. And button remapping is not supported on all devices currently.

hxss, 2018-05-20:
Maybe you know - if this mouse generate mouse button click event by dpi button?
I think if yes, then I can use xbindkeys for remapping and configuring modes.

hxss, 2018-05-20:
>And button remapping is not supported on all devices currently
Razer Orochi(Do I must create new issue for this question?) have 2 buttons on the right side that toggle dpi by default.
http://dl.razerzone.com/master-guides/Orochi2013/Orochi2013OMG-ENG.pdf

So we can't remap this buttons too?

z3ntu, 2018-05-20:
> Maybe you know - if this mouse generate mouse button click event by dpi button?

no idea

> So we can't remap this buttons too?

Don't think so. And don't create a new issue for that, there are a bunch of issues open for this thing right now anyways.

git-ingham, 2018-06-22:
I would like to add my voice for Atheris support.  Just bought it.  It works fine on my OpenSUSE Tumbleweed system without it, but the extra features would be nice.

hxss, 2018-06-22:
@git-ingham did you try use it over bluetooth? I have problem with polling rate when use this mouse over bt.
https://www.reddit.com/r/linuxquestions/comments/8qviub/bluetooth_mouse_polling_rate/

interisti, 2018-06-22:
@git-ingham 
trying to make it work on ubuntu 16.0.4, with no luck so far. System does not recognize it at all. Which package have you installed?

git-ingham, 2018-06-22:
It arrived today, I paired it, and it worked.  Just like that.  I have more problems with my Orochi than the Atheris.

hxss, 2018-06-23:
@git-ingham what rate shows evhz over bt?

nbenitez, 2018-09-02:
+1 for Razer Atheris, please add "Device Support" label.

jrbtaylor, 2018-09-14:
I would also like to request support for the Atheris

ayush2808, 2018-12-26:
I'm a relatively new user of the Atheris. The only workaround I've found with this if you have a Windows/Mac machine side by side. Whether it works from a different machine or not I do not know. But I have dual-booted Windows and Ubuntu 18.04.

Basically, I created a new profile on the Razer Synapse software and assigned **Ubuntu keyboard shortcuts** to the configurable keys and it works like a charm. The only things that don't work are the Hypreshift functions of Button 4,5 i.e. the side buttons. The other functions work like a charm.

Ex. I have Ctrl+Tab and Crl+Shift+Tab for the side buttons(standard) and shift workspaces with the scroll wheel with the hypershift key which is the middle button in my case. I have Copy and Paste as hypershift functions for the side buttons but they don't work.

Key thing is to activate that profile before logging out of windows and then start Ubuntu. I'm in USB mode and not Bluetooth. Don't have Bluetooth on my machine. If anybody finds any workarounds keep updating this thread.

Cheers!

cheuksing, 2019-06-29:
When I using Atheris in linux (pop os), the tracking is unstable.
I found that the mouse will fail to track for 0.1s if it stops moving and moved again after 5s.
I changed poll rate to 1000hz and check the poll rate in evhz.
This is what I got. Any idea?

Hopefully this project will support this mice and solve this issue.  :/

> Razer Razer Atheris - Mobile Gaming Mouse: Latest    76Hz, Average     0Hz
> Razer Razer Atheris - Mobile Gaming Mouse: Latest   100Hz, Average    38Hz
> Razer Razer Atheris - Mobile Gaming Mouse: Latest   111Hz, Average    58Hz
> Razer Razer Atheris - Mobile Gaming Mouse: Latest  1000Hz, Average    71Hz
> Razer Razer Atheris - Mobile Gaming Mouse: Latest    35Hz, Average   257Hz
> Razer Razer Atheris - Mobile Gaming Mouse: Latest    30Hz, Average   220Hz
> Razer Razer Atheris - Mobile Gaming Mouse: Latest    26Hz, Average   193Hz
> Razer Razer Atheris - Mobile Gaming Mouse: Latest     1Hz, Average   172Hz
> Razer Razer Atheris - Mobile Gaming Mouse: Latest  1000Hz, Average   153Hz
> Razer Razer Atheris - Mobile Gaming Mouse: Latest   500Hz, Average   237Hz

kreaninw, 2019-08-05:
I am about to buy this mouse to use with Ubuntu 19.04. I hope this model will be support soon.

z3ntu, 2019-08-08:
Can somebody give me the output of `lsusb | grep 1532` ?

mustafirus, 2019-08-14:
lsusb | grep 1532 
Bus 001 Device 005: ID 1532:0062 Razer USA, Ltd

nbenitez, 2019-08-17:
Mine is:
`Bus 001 Device 061: ID 1532:0227 Razer USA, Ltd Razer Huntsman`

git-ingham, 2019-08-17:
Here is mine:

Bus 001 Device 002: ID 1532:0048 Razer USA, Ltd Orochi (Wired)

Kenneth

fabiocarneiro, 2020-01-16:
```
Bus 001 Device 005: ID 1532:0232 Razer USA, Ltd Razer Blade Stealth
Bus 001 Device 007: ID 1532:0062 Razer USA, Ltd Razer Atheris - Mobile Gaming Mouse
```

Scarletvye, 2020-07-01:
Bus 001 Device 004: ID 1532:0062 Razer USA, Ltd

I have 2 of these devices, please let me know how I can help. I am running Linux Mint 20. I really want to get this mouse up to its 1000 polling rate.

fabiocarneiro, 2020-07-02:
@Scarletvye The Atheris works fine without any OpenRazer support, so the main thing for me would be to be able to visualize the DPI settings. I have some jumping sometimes, but I suspect it's not related to the mouse itself.

nightsky30, 2020-07-02:
@Scarletvye @crian has an Atheris [branch](https://github.com/crian/openrazer/tree/feature_atheris). Not sure what state that branch is in, but you could try that.

Belthil, 2020-07-09:
I guess this might be an ignorant question, but how can I use the atheris branch? I have installed openrazer from terminal on Pop_OS and RazerGenie, but no idea how to implement the Atheris branch to my installation...

nightsky30, 2020-07-10:
You'd have to clone and build it. I think Pop_OS should be able to follow the Ubuntu/Mint [instructions](https://github.com/openrazer/openrazer/wiki/Building-a-package). Just insert crains fork/branch in the clone command and follow the rest.

```git clone git@github.com:crian/openrazer.git -b feature_atheris```

Belthil, 2020-07-10:
Thanks for the reply and instructions!

crian, 2020-07-10:
This branch is old and not working as expected.

nightsky30, 2020-07-18:
@Belthil @crian Feel free to try my PR branch. I think I ironed out the issues. It is currently working well for me :smiley:

nightsky30, 2020-07-19:
@fabiocarneiro I'm also seeing some jumping/stuttering sometimes. Not sure if it is openrazer related or not. Seems I can recreate by viewing a web page and scroll wheeling a bit, then stopping the scroll wheel and moving the mouse about. Is this what you're seeing without openrazer?

Belthil, 2020-07-19:
@nightsky30 I saw the same jumping, or rather, lazy response. In all honesty, the atheris works fine without the openrazer drivers. My hope was to remap it's buttons or create macros on it. I managed to do that by using xbindkeys, but it was only partial success.

nightsky30, 2020-07-19:
> @nightsky30 I saw the same jumping, or rather, lazy response. In all honesty, the atheris works fine without the openrazer drivers. My hope was to remap it's buttons or create macros on it. I managed to do that by using xbindkeys, but it was only partial success.

It seems the issue is referenced on Razer's support page. There is also a firmware update available for Windows. If I can get my VM up and running again, I may try to update the FW through the USB pass-through and report back. My Atheris is currently running v1.3.

closed 2020-09-26 by z3ntu