openrazer/openrazer #599: "segmentation fault" when starting daemon with blade stealth

mingyech, 2018-05-27:
I am getting a "segmentation fault" when trying to start the daemon. 
I have a Blade stealth early 2016 (1532:0205) and Mamba chroma wireless (1532:0045)
I am running Arch linux, kernel 4.16.11-1-ARCH
I am already inside the plugdev group

> openrazer-daemon -Fv
Starting daemon.
2018-05-27 17:18:18 | razer                          | INFO     | Initialising Daemon (v2.3.0). Pid: 2950
2018-05-27 17:18:18 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderElite -------------------- (1532:005C)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerExpert ----------------- (1532:0202)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerFirefly ---------------------------- (1532:0C00)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerImperator -------------------------- (1532:002F)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerKrakenClassic ---------------------- (1532:0501)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerMambaTE ---------------------------- (1532:0046)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerNagaChroma ------------------------- (1532:0053)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaverChroma -------------------- (1532:0207)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerOrochi2011 ------------------------- (1532:0013)
[... cut ...]
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0201)
2018-05-27 17:18:18 | razer                          | DEBUG    | Loaded device specification: RazerTartarusChroma --------------------- (1532:0208)
2018-05-27 17:18:18 | razer                          | INFO     | Found device.0: 0003:1532:0205.0006
2018-05-27 17:18:18 | razer.device0                  | INFO     | Initialising device.0 RazerBladeStealth
Segmentation fault

Please help me! thanks.

z3ntu, 2018-05-27:
Can you run
```
gdb /usr/bin/python3
(gdb) run /usr/bin/openrazer-daemon -Fv
```
and when it crashes type
```
bt
```
to get the backtrace? This very much looks like a bug in either some library we use or python itself. You will probably also need to compile python with debug symbols - see https://wiki.archlinux.org/index.php/Debug_-_Getting_Traces

mingyech, 2018-05-27:
Ok so I don't know if I did this correctly but I did:

> gdb /usr/bin/python3
GNU gdb (GDB) 8.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/python3...(no debugging symbols found)...done.
(gdb) run /usr/bin/openrazer-daemon -Fv
Starting program: /usr/bin/python3 /usr/bin/openrazer-daemon -Fv
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Starting daemon.
2018-05-27 18:46:56 | razer                          | INFO     | Initialising Daemon (v2.3.0). Pid: 3399
2018-05-27 18:46:56 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2018-05-27 18:46:56 | razer                          | DEBUG    | Loaded device specification: RazerDeathAdderChroma ------------------- (1532:0043)
[... cut ...]
    from openrazer_daemon.daemon import RazerDaemon, __version__
ImportError: No module named 'openrazer_daemon'
[Inferior 1 (process 10709) exited with code 01]
(gdb) set logging off
Done logging to trace3.log.
(gdb) quit

> cat trace3.log 
Starting program: /usr/bin/python3.5 /usr/bin/openrazer-daemon -Fv
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Inferior 1 (process 10709) exited with code 01]

 Let me know what I did wrong, and anythng else you want
 Thanks!

z3ntu, 2018-05-27:
Actually I think the output of the following statement is more useful: `python3 -m trace --trace /usr/bin/openrazer-daemon -Fv`
Put the output either in a text file and upload it here, or use pastebin or a similar service

z3ntu, 2018-05-27:
Did it crash that time? Also does python crash when executing `python3.5 /usr/bin/openrazer-daemon -Fv` (as you already have python3.5 installed :smiley: )

mingyech, 2018-05-27:
Yea it did, sorry after checking it seems like the text file does not contain all of the outputs, I did

> python3 -m trace --trace /usr/bin/openrazer-daemon -Fv > trace.txt

but it doesn't save the whole thing, here is the last part, copied from my terminal emulator:
[https://pastebin.com/Yw2S5dFC](https://pastebin.com/Yw2S5dFC)

python3.5 gives a different error:

> python3.5 /usr/bin/openrazer-daemon -Fv
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 11, in <module>
    from openrazer_daemon.daemon import RazerDaemon, \_\_version\_\_
ImportError: No module named 'openrazer_daemon'

Thanks!:D

z3ntu, 2018-05-27:
Yeah that python3.5 package seems to use a different prefix than the normal python package.
Okay `ls -al /sys/bus/hid/drivers/razerkbd/0*/device_serial`, `cat /sys/bus/hid/drivers/razerkbd/0*/device_serial` and `python3 -c "print(open('"$(realpath /sys/bus/hid/drivers/razerkbd/0*/device_serial)"').read())"` please.

mingyech, 2018-05-27:
> ls -al /sys/bus/hid/drivers/razerkbd/0*/device_serial
-r--r----- 1 root plugdev 4096 May 27 16:49 /sys/bus/hid/drivers/razerkbd/0003:1532:0205.0006/device_serial

> cat /sys/bus/hid/drivers/razerkbd/0*/device_serial
Segmentation fault

> python3 -c "print(open('"$(realpath /sys/bus/hid/drivers/razerkbd/0*/device_serial)"').read())"
Segmentation fault

z3ntu, 2018-05-27:
Wait, even `cat` is having a segmentation fault?? Try rebooting maybe?

mingyech, 2018-05-27:
Rebooted, then I got this at login:
[https://pasteboard.co/HnaCCpq.jpg](https://pasteboard.co/HnaCCpq.jpg)
Never happend before, cat works with other stuff, I tried to open it: 
[https://pasteboard.co/HnaD1sk.jpg](https://pasteboard.co/HnaD1sk.jpg)
Do you think its something messed up in my laptop? I just installed arch yesturday...
Btw I'm gonna be afk for a while, Thanks for your help!

mingyech, 2018-05-27:
And yes it still crashes

z3ntu, 2018-05-27:
Doesn't look good :/ Try reinstalling the kernel package? And is there anything new in the dmesg when you run the `cat /sys/bus/hid/drivers/razerkbd/0*/device_serial` ?

z3ntu, 2018-05-27:
I just realized that message is caused by our driver (in the Call Trace there's our azer_attr_read_get_serial` function. Can you run `sudo dmidecode -t 1` and give me the output?

mingyech, 2018-05-28:
> sudo dmidecode -t 1
\# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.
\
Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Razer
	Product Name: Blade Stealth
	Version: 1.04
	Serial Number: As printed in the D cover
	UUID: blablabla
	Wake-up Type: Power Switch
	SKU Number:  
	Family:

z3ntu, 2018-05-28:
What I expected, the serial string (25 chars) is larger than our data structure (23 chars).

commit 724acdac (2018-05-28): razerkbd: Increase size of serial_string variable and use strncpy

Due to beautiful malformed (and too long) serial strings on Blade laptops and the
strcpy function, there was a buffer overflow.

Thanks @terrycain for the advice :)

Fixes #599
closed 2018-05-28 by z3ntu