openrazer/openrazer #601: Not detecting my Kraken Headset

bertokhoury, 2018-05-28:
All drivers are up to date, but my device is not being recognized. Must the headset device be USB or audio jack?

Danonk, 2018-05-29:
Hello, I have DeathAdder mouse and Kraken Headset. RazerGenie detects my mouse, but does't detect my headset.

The command: `lsusb | grep 1532` displays those:
```
danon@falcon:~$ lsusb | grep '1532:'
Bus 001 Device 014: ID 1532:0502 Razer USA, Ltd 
Bus 001 Device 015: ID 1532:005c Razer USA, Ltd
```

`1532:005c` is the mouse (as I've read from your ReadMe.md file) and I know that `1532:0501` is Kraken 7.1, but I see `0502` (which is also a Kraken Headset, but probably an unsupported one?).

Why doesn't RazerGenie doesn't support my `1532:0502' headphones?

bertokhoury, 2018-05-29:
Yes @Danonk  it is the Kraken 7.1 which I believe is an unsupported headset if you have had the same issue. 
Thank you for the command @z3ntu . 

I am closing this question now.

closed 2018-05-29 by bertokhoury

z3ntu, 2018-05-29:
@Danonk  Does the headset have LEDs you can configure on Windows?

Danonk, 2018-06-01:
@bertokhoury You didn't answer my question, why isn't that particular device supported?

@z3ntu Didn't check yet