openrazer/openrazer #602: razer blade stealth late 2017 doesn't detect keyboard

edvorg, 2018-05-29:
Hi. Came across openrazer, but couldn't make it work even though my device is in the list of supported devices
```
[edvorg@edvorg-manjaro ~]$ lsusb | grep 1532
Bus 001 Device 005: ID 1532:0232 Razer USA, Ltd 
[edvorg@edvorg-manjaro ~]$ uname -a
Linux edvorg-manjaro 4.14.40-1-MANJARO #1 SMP PREEMPT Wed May 9 20:10:25 UTC 2018 x86_64 GNU/Linux
```
tried razer commander and polychromatic, both tell me they can connect to openrazer but there are no connected razer devices in the system

z3ntu, 2018-05-29:
Please post the output of both `sudo dkms install openrazer-driver/1.0.0` and `openrazer-daemon -Fv`

edvorg, 2018-05-29:
Hello again. I apologize for posting the issue. It got fixed after reboot.
FYI `sudo dkms install openrazer-driver/1.0.0` successfully installs module
probably something was preventing functionality to detect the device

closed 2018-05-29 by edvorg