openrazer/openrazer #604: Sometimes brightness set to and remains at 0% when returning from suspend

nightsky30, 2018-06-02:
BlackWidow Chroma v2 sometimes has brightness set to and remains at 0% when returning from suspend.  Other times the keyboard correctly has its brightness restored.

When the brightness is left at 0% I can enter the tray applet and manually change the brightness back to 100%.

~$ uname -a
Linux 4.15.0-22-generic #24-Ubuntu SMP Wed May 16 12:15:17 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04 LTS
Release:	18.04
Codename:	bionic

muesli, 2018-07-15:
I experience the same issue on ArchLinux with openrazer `2.3.1` and Linux `4.17.5-1`.

nightsky30, 2018-07-15:
@muesli Also BW Chroma V2 ??

muesli, 2018-07-15:
@nightsky30 No, a Ornata Chroma.

IvanVojtko, 2018-10-05:
Same issue for me on Arch Linux with Ornata Chroma.

muesli, 2018-10-22:
Just caught the following output from dbus calls, after the keyboard went dark:

```
2018/10/21 12:56:05 dbus call failed: Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py", line 490, in set_key_row
    with open(driver_path, 'wb') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:021E.0003/matrix_custom_frame'
2018/10/21 12:56:05 dbus call failed: Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py", line 490, in set_key_row
    with open(driver_path, 'wb') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:021E.0003/matrix_custom_frame'
2018/10/21 12:56:05 dbus call failed: Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py", line 490, in set_key_row
    with open(driver_path, 'wb') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:021E.0003/matrix_custom_frame'
2018/10/21 12:56:05 dbus call failed: Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py", line 490, in set_key_row
    with open(driver_path, 'wb') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:021E.0003/matrix_custom_frame'
2018/10/21 12:56:05 dbus call failed: Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py", line 490, in set_key_row
    with open(driver_path, 'wb') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:021E.0003/matrix_custom_frame'
2018/10/21 12:56:05 dbus call failed: Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py", line 490, in set_key_row
    with open(driver_path, 'wb') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:021E.0003/matrix_custom_frame'
2018/10/21 12:56:05 dbus call failed: Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3.7/site-packages/openrazer_daemon/dbus_services/dbus_methods/chroma_keyboard.py", line 466, in set_custom_effect
    with open(driver_path, 'wb') as driver_file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:021E.0003/matrix_effect_custom'
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setCustom" with signature "" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setKeyRow" with signature "ay" on interface "razer.device.lighting.chroma" doesn't exist
2018/10/21 12:56:06 dbus call failed: Method "setCustom" with signature "" on interface "razer.device.lighting.chroma" doesn't exist
```

redlion57, 2018-11-24:
hi all, 
I've got the same issue, sometimes after a reboot or wake up the brightness is set automatically at 0%.
Device : Razer MAMBA tournament edition
Thanks.

$ uname -a
Linux parrot 4.18.0-parrot10-amd64 #1 SMP Debian 4.18.10-2parrot10 (2018-11-17) x86_64 GNU/Linux

$lsb_release -a
No LSB modules are available.
Distributor ID:	Parrot
Description:Parrot 4.4
Release:	4.4
Codename:stable

z3ntu, 2018-11-25:
@muesli The stacktrace you posted was fixed by https://github.com/openrazer/openrazer/commit/00a665a7514f4a20c8f8c4d98bf0a0fd4826af10 (released in v2.4.0)

I'm gonna close this as a duplicate of #525 (which seems to be the root cause for this)

closed 2018-11-25 by z3ntu